react-native-camera/react-native-camera #340: Barcode EAN13 does not scan when presented horizontally

ashwini-c, 2016-07-13:
### Steps to reproduce
1. Try to scan barcode EAN13 format
2. Present barcode horizontally (does not scan easily)
3.  Present barcode vertically (Scans quickly)

### Expected behaviour
Should scan both horizontally and vertically

### Actual behaviour
 Present barcode horizontally (does not scan easily)
 Present barcode vertically (Scans quickly)

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: 0.3.8

E-P1983, 2016-08-01:
Can confirm this bug.

alvarowolfx, 2016-09-06:
I'm experiencing this bug on iOS , I want to help on this but don't even know what causes this.

E-P1983, 2017-03-15:
I've tried to track it down in the code, but have not found anything useful yet.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco