react-native-camera/react-native-camera #341: When I disabled camera

Lepinr, 2016-07-13:
### Steps to reproduce
1. First time to open the camera on ios
2. select denied permission
3. app has crashed...

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**: 5.8
- **React Native version**: 0.26.2
- **React Native platform + platform version**: iOS 8.4

### react-native-camera
**Version**: "master"

## Xcode logs
2016-07-14 04:02:54.435 [info][tid:main][RCTCameraManager.m:244] Error setting property 'captureAudio' of RCTCamera with tag #1317: capturing audio
2016-07-14 04:02:54.577 yungou[8532:2213124] Error Domain=AVFoundationErrorDomain Code=-11852 "Cannot use Back Camera" UserInfo=0x3d4b070 {NSLocalizedDescription=Cannot use Back Camera, AVErrorDeviceKey=<AVCaptureFigVideoDevice: 0x3d56ff0 [Back Camera][com.apple.avfoundation.avcapturedevice.built-in_video:0]>, NSLocalizedFailureReason=This app is not authorized to use Back Camera.}
2016-07-14 04:02:54.583 yungou[8532:2213124] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '*** -[AVCaptureMetadataOutput setMetadataObjectTypes:] - unsupported type found.  Use -availableMetadataObjectTypes.'
*** First throw call stack:
(0x299e0fef 0x38324c8b 0x2855c4b1 0x53a3d 0x388b52e3 0x388bd729 0x388b7aad 0x388bef9f 0x388c03c3 0x38a1cdc1 0x38a1cb14)
libc++abi.dylib: terminating with uncaught exception of type NSException

Lepinr, 2016-07-13:
我想用户们也可能会禁止相机访问权限，如果出现闪退会使他们非常不满意。我目前还不是太懂Objective-C语言。
---------------------------
I think users may also prohibit camera access, if there is a flash back will make them very dissatisfied. I am not too understand Objective-C language.

hilkeheremans, 2016-07-18:
See #134

Lepinr, 2016-07-28:
Thank you @hilkeheremans

closed 2016-07-28 by Lepinr