react-native-camera/react-native-camera #342: CameraManager.CaptureQuality.high

dunglt90vt, 2016-07-14:
###ISSUE
I got an issue :
undefined is not an object (evaluating 'CameraManager.CaptureQuality.high')
unhandled JS Exception: undefined is not an object (evaluating 'CameraManager.CaptureQuality.high')
Module AppRegistry is not a registered callable module.
Unhandled JS Exception: Module AppRegistry is not a registered callable module.

### Environment
- **Node.js version**: CameraManager.CaptureQuality.high
- **React Native version**: 0.28
- **React Native platform + platform version**: iOS 9.+, iOS simulator 9.3

### react-native-camera
**Version**: npm version 2.15.8

Please show me how to solve this! Thanks

dunglt90vt, 2016-07-15:
Solved. I remove this line :)

closed 2016-07-15 by dunglt90vt

chenkehxx, 2016-07-23:
@dunglt90vt  how do you solve it?