react-native-camera/react-native-camera #343: data.path is always null

Thomsos, 2016-07-14:
### Steps to reproduce
1. Press button to call capture method

### Expected behaviour
Return {path: 'file\\...."}

### Actual behaviour
Path is null.

### Environment
- **Node.js version**: 6.2.2
- **React Native version**: 0.28.0
- **React Native platform + platform version**:  Android 6.0

### react-native-camera
**Version**:  0.3.8

Thomsos, 2016-07-18:
Solved my own problem by adding:

`<uses-permission android:name="android.permission.CAMERA" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
  <uses-feature android:name="android.hardware.camera" android:required="false"/>
  <uses-feature android:name="android.hardware.camera.autofocus" android:required="false"/>`

To:  `android/app/src/main/AndroidManifest.xml `

I would recommend updating the docs to show this in the manual install.

closed 2016-07-18 by Thomsos

anton6, 2016-07-30:
In addition I also had to add: captureTarget={Camera.constants.CaptureTarget.disk} to Camera component.

mike-aungsan, 2017-07-12:
I have the same problem on Android 4.4, Device S4. Any help?