react-native-camera/react-native-camera #345: Feature Request - JPG from MOV

jackread, 2016-07-16:
When I shoot a video, it would be awesome to be able to pull a JPG frame from the video for previewing etc.

Is that something you would consider supporting?

ivanbrens, 2016-07-22:
+1, would be really cool to see this.  I have realized that the react-native ecosystem really leads you to this repo as the main camera solution.  It would be amazing to see features like this.  There's a really cool [camera repo](https://github.com/maxs15/react-native-screcorder) that was ported that supports this, but there is zero android support.

JLLLinn, 2017-12-24:
+1, this would be really useful

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco