react-native-camera/react-native-camera #346: Apps unexpectedly quit

StevenKing9, 2016-07-19:
### Steps to reproduce
1.same this ： https://github.com/lwansbrough/react-native-camera 
2.click the <text>
3.Apps unexpectedly quit

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:v6.3.0
- **React Native version**:0.29.2
- **React Native platform + platform version**:  Android 5.1

### react-native-camera
**Version**: npm version 3.8（ react-native-camera@0.3.8 ）

StevenKing9, 2016-07-19:
> msg's executing time is too long
07-19 22:08:20.505 16403-16403/com.test4 E/ANR_LOG: Blocked msg = { when=-1s848ms what=256 target=android.hardware.Camera$EventHandler obj=[B@1143ae63 } , cost  = 1848 ms
07-19 22:08:20.505 16403-16403/com.test4 E/ANR_LOG: >>>Current msg List is:
07-19 22:08:20.505 16403-16403/com.test4 E/ANR_LOG: Current msg <1>  = { when=-1s589ms what=0 target=android.graphics.SurfaceTexture$1 }
07-19 22:08:20.505 16403-16403/com.test4 E/ANR_LOG: Current msg <2>  = { when=-1s235ms what=0 target=android.os.Handler callback=com.facebook.react.devsupport.DevServerHelper$4$1 }
07-19 22:08:20.505 16403-16403/com.test4 E/ANR_LOG: Current msg <3>  = { when=0 what=131 target=android.app.ActivityThread$H obj=android.app.ActivityThread$ProviderRefCount@3c859a60 }
07-19 22:08:20.505 16403-16403/com.test4 E/ANR_LOG: >>>CURRENT MSG DUMP OVER<<<

StevenKing9, 2016-07-20:
This is my android/app/src/main/AndroidManifest.xml : 
 <uses-permission android:name="android.permission.MOUNT_UNMOUNT_FILESYSTEMS" />
 #    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_SETTINGS" />
    <uses-permission android:name="androi d.permission.WRITE_CONTACTS" />
    <uses-permission android:name="android.permission.READ_PHONE_STATE" />

    <permission android:name="android.permission.WRITE_MEDIA_STORAGE" />

    <uses-permission android:name="android.permission.RECORD_AUDIO" />
    <uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />

    <permission android:name="android.permission.DEVICE_POWER" />

    <uses-permission android:name="android.permission.WAKE_LOCK" />
    <uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-feature android:name="android.hardware.camera" android:required="false" />
    <uses-feature android:name="android.hardware.camera.autofocus" android:required="false" />

sibelius, 2018-01-09:
this looks like a permission issue

closed 2018-01-09 by sibelius