react-native-camera/react-native-camera #347: (iOS) barCodeTypes default not setting for types interleaved2of5, itf14, datamatrix

micksabox, 2016-07-20:
### Steps to reproduce
1. Integrate react-native-camera in your project, leaving barCodeTypes empty (should default to all available)
2. Set a breakpoint on RCTCameraManager.m didCaptureOutput:didOutputMetadataObjects:fromConnection:
3. Run the app on an iOS 8+ device and scan a datamatrix (http://stackoverflow.com/questions/27580953/unable-to-read-white-on-black-data-matrix-barcode) and then inspect self.barCodeTypes from within the breakpoint

### Expected behaviour
`po self.barCodeTypes` should print out all the barCodeTypes supported, which should include the interleaved2of5, itf14, datamatrix types

### Actual behaviour
These types are missing. 

### How to fix
The current implementation checks the available types using 
```
#ifdef AVMetadataObjectTypeInterleaved2of5Code
,@"interleaved2of5": AVMetadataObjectTypeInterleaved2of5Code
# endif
#ifdef AVMetadataObjectTypeITF14Code
,@"itf14": AVMetadataObjectTypeITF14Code
# endif
#ifdef AVMetadataObjectTypeDataMatrixCode
,@"datamatrix": AVMetadataObjectTypeDataMatrixCode
# endif
```
in RCTCameraManager.m -constantsToExport. I found that this worked for me:
```
NSMutableDictionary * barCodeTypes = [NSMutableDictionary dictionary];
    [barCodeTypes setDictionary:@{
    @"upce": AVMetadataObjectTypeUPCECode,
    @"code39": AVMetadataObjectTypeCode39Code,
    @"code39mod43": AVMetadataObjectTypeCode39Mod43Code,
    @"ean13": AVMetadataObjectTypeEAN13Code,
    @"ean8":  AVMetadataObjectTypeEAN8Code,
    @"code93": AVMetadataObjectTypeCode93Code,
    @"code138": AVMetadataObjectTypeCode128Code,
    @"pdf417": AVMetadataObjectTypePDF417Code,
    @"qr": AVMetadataObjectTypeQRCode,
    @"aztec": AVMetadataObjectTypeAztecCode
    }];
    //Check at runtime if these other constants exist
    if(&AVMetadataObjectTypeInterleaved2of5Code != nil){
        
        [barCodeTypes setObject:AVMetadataObjectTypeInterleaved2of5Code forKey:@"interleaved2of5"];
    }
    
    if(&AVMetadataObjectTypeITF14Code != nil){
        
        [barCodeTypes setObject:AVMetadataObjectTypeITF14Code forKey:@"itf14"];
    }
    
    if(&AVMetadataObjectTypeDataMatrixCode != nil){
        
        [barCodeTypes setObject:AVMetadataObjectTypeDataMatrixCode forKey:@"datamatrix"];
    }
```
but I'm not sure how that will work on an iOS 7 device.

### Environment
- **Node version**: 5.0.0
- **React Native version**:0.29.1
- **Platforms**: RN 0.29.1 + iOS 9.3.2

### react-native-camera
**Version**: git+https://github.com/lwansbrough/react-native-camera.git

LeBlaaanc, 2016-09-13:
What do we need to do to get this fixed in the current branch? Test iOS 7 with the change?

micksabox, 2016-09-13:
@LeBlaaanc or be sure that it won't cause a crash. I never used this `&AVMetadataObjectTypeDataMatrixCode` to check existence of a constant at runtime, so I can't say much about if that will cause a runtime crash.

Another option is to just ensure you're deploy target is greater than X, where the iOS SDK in version X contains the constant your checking against.

LeBlaaanc, 2016-09-13:
@micksabox right, I just was curious how we get this into a PR, we target iOS 9+ and it works fine.

LeBlaaanc, 2016-09-16:
I think I spoke too soon @micksabox did you have issues with this only in production? Ours reader picks up interleaved2of5 until it's compiled.

micksabox, 2016-09-16:
Hey @LeBlaaanc I'm not sure, I found this issue in dev, haven't tried in production.

robwalkerco, 2016-11-17:
Can we get this changed to use `&AVMetadataObjectTypeDataMatrixCode` etc. now?
Not sure what level of backwards compatibility is required for iOS 7 now?

robwalkerco, 2016-11-17:
I see there is already a PR at #401 that works well for me 👍

msom, 2016-12-01:
#401 works for me as well

ptomasroos, 2017-02-13:
This is now merged into master with #586 https://github.com/lwansbrough/react-native-camera/commit/371a6a0dfaec1aa744087e3ceb0deb404055b189

closed 2017-02-13 by ptomasroos