react-native-camera/react-native-camera #350: Video camera is working, but the data is not being returned

hermant2, 2016-07-21:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

So I have a component that takes videos, but when I try to `console.log` the data, nothing is happening.  I am sure that the camera is taking videos though because when I set the `captureTarget` to `cameraRoll

### Actual behaviour
Tell us what happens instead

### Environment
- **Node.js version**:
- **React Native version**:
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: npm version or "master"

RichLewis007, 2016-07-28:
I'm also trying to get video to work. I'm working from the Example code of the project, and have tried adding captureMode: Camera.constants.CaptureMode.video to the camera state, but haven't gotten anything but picture taking so far.

krillov, 2016-08-23:
@RichLewis007 Any luck with the video? I got the same problems. It takes pictures only.

bsiddiqui, 2016-08-24:
@krillov I was able to get the video working - mind positing your component and I'll try and help you out

krillov, 2016-08-24:
@bsiddiqui Thanks. I already have a working camera app and I wanted to implement video function also. I'm using iphone simulator in Xcode. It simulates taking pictures great but I'm not sure if it can also simulate recording video.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco