react-native-camera/react-native-camera #352: Captured image have wrong orientation on Samsung S5

markos85, 2016-07-29:
It seems that param orientation doesn't work at all on the Samsung S5 device. Whatever orientation value is passed to native, it doesn't work. The image is always in same, wrong orientation.

### Steps to reproduce
1. Capture image with Samsung S5

### Expected behaviour
Go to the recent/gallery and you will see that image have wrong orientation ( to the left)

### Actual behaviour
An image should be saved as it's captured, in portrait mode.

### Environment
- **React Native version**: 0.30
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: 0.3.8

jakegardner, 2016-07-29:
experiencing the same thing on ios sim/iphone.

charpeni, 2016-08-02:
Thank you for reporting.

I'll close this issue since it's a duplicate of #309.

closed 2016-08-02 by charpeni

npomfret, 2017-04-26:
@markos85  are you able to test my PR?  It may help with this https://github.com/lwansbrough/react-native-camera/pull/682.