react-native-camera/react-native-camera #353: Unable to run example

gonzaloserrano, 2016-07-31:
### Steps to reproduce
1. clone the repo
2. cd to `Example`
3. run `npm init`
4. run `npm run-ios`

### Expected behaviour
The app should be run in the simulator.

### Actual behaviour
The simulator shows an error.
[image]

### Environment
- **Node.js version**: v6.3.0
- **React Native version**: react-native-cli: 1.0.0 react-native: 0.26.3
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**: master

gonzaloserrano, 2016-07-31:
duplicated of https://github.com/lwansbrough/react-native-camera/issues/220

closed 2016-07-31 by gonzaloserrano

prabhaav, 2016-07-31:
This fixed the problem for me:

Change the first 2 lines in index.ios.js to:

import React, {PropTypes} from 'react'
import { StyleSheet, requireNativeComponent, NativeModules, DeviceEventEmitter } from 'react-native'

gonzaloserrano, 2016-07-31:
Not for me:

```
2016-07-31 17:48:11.590 [error][tid:com.facebook.react.JavaScript] undefined is not an object (evaluating 'CameraManager.CaptureQuality.high')
2016-07-31 17:48:11.592 [fatal][tid:com.facebook.react.RCTExceptionsManagerQueue] Unhandled JS Exception: undefined is not an object (evaluating 'CameraManager.CaptureQuality.high')
2016-07-31 17:48:12.666 [error][tid:com.facebook.react.JavaScript] Module AppRegistry is not a registered callable module.
2016-07-31 17:48:12.667 [fatal][tid:com.facebook.react.RCTExceptionsManagerQueue] Unhandled JS Exception: Module AppRegistry is not a registered callable module.
```

Thanks anyway!