react-native-camera/react-native-camera #357: Caused by: java.lang.RuntimeException: Camera is being used after Camera.release() was called

yanpeipan, 2016-08-02:
### Steps to reproduce
1. AView push to BView then pop to AView with Navigator, and BView has Camera
2. Repeated step 1 several times, It will crash with log:
```
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #7
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime: Process: com.rn_basic, PID: 26725
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime: java.lang.RuntimeException: An error occurred while executing doInBackground()
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at android.os.AsyncTask$3.done(AsyncTask.java:309)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:354)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.FutureTask.setException(FutureTask.java:223)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.FutureTask.run(FutureTask.java:242)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.lang.Thread.run(Thread.java:818)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:  Caused by: java.lang.RuntimeException: Camera is being used after Camera.release() was called
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at android.hardware.Camera.setHasPreviewCallback(Native Method)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at android.hardware.Camera.setOneShotPreviewCallback(Camera.java:953)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at com.lwansbrough.RCTCamera.RCTCameraViewFinder.onResult(RCTCameraViewFinder.java:166)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at com.lwansbrough.RCTCamera.BarcodeReader$ReaderAsyncTask.doInBackground(BarcodeReader.java:62)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at com.lwansbrough.RCTCamera.BarcodeReader$ReaderAsyncTask.doInBackground(BarcodeReader.java:40)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at android.os.AsyncTask$2.call(AsyncTask.java:295)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.FutureTask.run(FutureTask.java:237)
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:234) 
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
08-02 15:19:46.891 26725-26880/com.rn_basic E/AndroidRuntime:     at java.lang.Thread.run(Thread.java:818) 
```

### Environment
- **Node.js version**: 3.8.3
- **React Native version**: 0.30
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: master

raulrene, 2016-09-01:
I'm also getting this in an example I'm trying to build. Also happens on the [Example.js](https://github.com/lwansbrough/react-native-camera/blob/master/Example/Example.js) provided in the library.
Could it be something device-related?

I'm running with React Native 0.32, Android 6.0.1 on a real device.

Taking pictures works well and the video also starts, but when I press Stop button it just crashes with the `Camera is being used after Camera.release() was called` error. I can't seem to get to the bottom of it.

will-ockmore, 2016-09-13:
Also run into this issue - in our case we are using the camera component in a subview tab. If you switch the tabs too quickly, the app crashes with this error.

chrisdukey, 2016-09-13:
I have a fix for this, it's caused because the preview callback isn't set to null before the camera is released. Will try and push it out this evening.

adinin, 2016-11-30:
@chrisdukey Sounds like you are referring to [PR#410](https://github.com/lwansbrough/react-native-camera/pull/410/files). I built from the latest on master, with your PR and I'm still experiencing the error as described by @yanpeipan 
Maybe this has manifested over again as a result of some other changes.
Do you have any other ideas why this could be happening?

Thank you in advance!

chrisdukey, 2016-12-01:
@adinin Yep, that's my PR. I've just tried it out from the latest on master and I don't appear to have any issues. [This is the code I tested with](https://github.com/chrisdukey/testcamera2). I built with RN 0.33.0, Node 6.2.0 and tested on a real Android device (6.0.1). [Here's a little movie as proof!](https://drive.google.com/file/d/0B6-7Nh1ywaVQTXJGdENoUWFHNlk/view?usp=sharing). Can you try my code or share a small example that gives the exception for you so I can test?

adinin, 2016-12-02:
hey @chrisdukey 
I'm running this on RN 0.32, Node v6.5.0, Android 6.0.1

I was able to mitigate the situation by having Camera removed from my view before I transition away to a different screen (scanDone state below). 

Also, before this mitigation, the problem used to occur only after the second time I would transition away from a screen with a Camera in it. 

This is my current render function:

```
  render(){
    if(this.state.scanDone){
      return (
	<View style={{flex:1}}>
	  <ActivityIndicator
	    size="large"
	    animating={true}
	    style={styles.spinner}
	  />
	  <Text>{this.state.feedbackText}</Text>
	</View>
      );
    }else{
      return (
      <Camera
	ref={(cam) => {
	  this.camera = cam;
	}}
	style={styles.preview}
	onBarCodeRead={this.barcodeReceived}
	aspect={Camera.constants.Aspect.fill}>
	  <View style={styles.rectangleContainer}>
	    <View style={styles.cameraRectangle} />
	  </View>
      </Camera>
      );
    }
  },
```

gaykov, 2016-12-02:
I'm facing the same error when closing the app by clicking home system button. Seems like camera component still trying to work in background.

lukaleli, 2017-06-28:
Is this planned to be fixed? I'm still getting this error on RN0.41.2 with react-native-camera@0.8.0

emmajam, 2017-07-03:
Getting this error also - fix would be great 🙏

stoneman1, 2017-07-03:
Getting same error, but instead getting it when using the back button on android

jslok, 2017-10-11:
Same error. Happens when
navigate to camera screen
navigate to a second camera screen
crash

moshema, 2017-10-13:
+1 on this bug...

ThaJay, 2017-12-20:
navigate to camera screen
scan barcode - display modal
press back button
crash

This does not always happen, it can also navigate back like it's supposed to, but it does have a pretty high chance of happening.
```
E/AndroidRuntime(13470): FATAL EXCEPTION: AsyncTask #1
E/AndroidRuntime(13470): Process: com.chipta, PID: 13470
E/AndroidRuntime(13470): java.lang.RuntimeException: An error occured while executing doInBackground()
E/AndroidRuntime(13470): 	at android.os.AsyncTask$3.done(AsyncTask.java:300)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:355)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.setException(FutureTask.java:222)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.run(FutureTask.java:242)
E/AndroidRuntime(13470): 	at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:231)
E/AndroidRuntime(13470): 	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1112)
E/AndroidRuntime(13470): 	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:587)
E/AndroidRuntime(13470): 	at java.lang.Thread.run(Thread.java:841)
E/AndroidRuntime(13470): Caused by: java.lang.RuntimeException: Method called after release()
E/AndroidRuntime(13470): 	at android.hardware.Camera.native_getParameters(Native Method)
E/AndroidRuntime(13470): 	at android.hardware.Camera.getParameters(Camera.java:1765)
E/AndroidRuntime(13470): 	at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:309)
E/AndroidRuntime(13470): 	at com.lwansbrough.RCTCamera.RCTCameraViewFinder$ReaderAsyncTask.doInBackground(RCTCameraViewFinder.java:294)
E/AndroidRuntime(13470): 	at android.os.AsyncTask$2.call(AsyncTask.java:288)
E/AndroidRuntime(13470): 	at java.util.concurrent.FutureTask.run(FutureTask.java:237)
E/AndroidRuntime(13470): 	... 4 more
```

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

ThaJay, 2018-01-16:
This is a critical issue! It is relevant now and needs to be solved.

jgcmarins, 2018-01-16:
@ThaJay is it happening on 0.13.0?

ThaJay, 2018-01-16:
Sorry, I have not checked yet as I am busy with another fix atm. But I'll check now if I can do it quickly.

emmajam, 2018-01-16:
@ThaJay and @jgcmarins yes it still occurs in 0.13.0

emmajam, 2018-01-16:
It is definitely a highly critical issue, our company is building its own RN scanning component as we can't rely on this one *mostly* due to this issue and its lack of attention ... plus others 😢

sibelius, 2018-01-17:
@emmajam can you try experimental RNCamera?

join us on expo slack on #react-native-camera channel, let's see if we can help

rf1804, 2018-03-10:
I am getting this error when video recording is started and i press the back button of device without stopping the video, any solution for it?

ZionChang, 2018-03-16:
Also crash in RNCamera

shahchaitanya, 2018-08-02:
@sibelius Yes it is happening in Rn Camera. Stacktrace is already mentioned by @ThaJay

sibelius, 2018-08-02:
the stacktrace is for RCTCamera, we need a stacktrace for RNCamera

shahchaitanya, 2018-08-03:
Here it's stacktrace.
`Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void org.reactnative.facedetector.RNFaceDetector.release()' on a null object reference
        at org.reactnative.camera.RNCameraView.onHostDestroy(RNCameraView.java:458)
        at com.facebook.react.bridge.ReactContext.onHostDestroy(ReactContext.java:230)
        at com.facebook.react.ReactInstanceManager.moveToBeforeCreateLifecycleState(ReactInstanceManager.java:690)
        at com.facebook.react.ReactInstanceManager.onHostDestroy(ReactInstanceManager.java:599)
        at com.facebook.react.ReactInstanceManager.onHostDestroy(ReactInstanceManager.java:613)
        at com.facebook.react.ReactActivityDelegate.onDestroy(ReactActivityDelegate.java:119)
        at com.facebook.react.ReactActivity.onDestroy(ReactActivity.java:72)
        at android.app.Activity.performDestroy(Activity.java:7209)
        at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1265)
        at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:4629)
        at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:4660) 
        at android.app.ActivityThread.-wrap5(Unknown Source:0) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1749) 
        at android.os.Handler.dispatchMessage(Handler.java:105) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6809)`

SimonErm, 2018-09-25:
Fixed in https://github.com/react-native-community/react-native-camera/commit/71a535571e4d06d8368bd46987f3878831a14c07

closed 2018-09-25 by SimonErm