react-native-camera/react-native-camera #359: Permission Denied while trying to save captured Image: RCTCameraModule

RichardBoyewa, 2016-08-05:
### Steps to reproduce
1. After installing the camera plugin
2. make sure your build.gradle classpath is ```classpath 'com.android.tools.build:gradle:1.2.3'```
3. Run the app on your device
4. But if you keep an higher version of build.gradle class path it works fine : ```classpath 'com.android.tools.build:gradle:1.3.1' ```

### Expected behaviour
Should capture the image and return a data object that has path property

### Actual behaviour
Displays a red screen showing this message
``` the callback capture() exist in module rctcamera, but only one callback may be registered to a function in a native module ```

### Environment
- **React Native version**: 0.30.0
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc

### react-native-camera
**Version**: react-native-camera@0.3.8

charpeni, 2016-08-28:
Can you test with master instead of 0.3.8 from npm ?

`npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save`

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco