react-native-camera/react-native-camera #361: Camera.checkDeviceAuthorizationStatus is not a function

EverChris, 2016-08-09:
[image]

why?

EverChris, 2016-08-09:
help me

markomatic, 2016-08-11:
@EverChris, are you calling it statically on imported Camera class?

import Camera from 'react-native-camera';
...
Camera.checkDeviceAuthorizationStatus()...

realjamie, 2016-09-08:
I had the same issue. My react-native-camera was working well until I upgrade it to 0.3.8 and upgrade ReactNative to 0.32.0

[Environment]
Node.js version: 4.2.4
React Native version: 0.32.0
React Native platform + platform version: iOS

[react-native-camera]
Version: 0.3.8

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco