react-native-camera/react-native-camera #362: Importing Camera causes "ReactNative.createClass" error

J-Gonzalez, 2016-08-09:
Running:
react: "15.2.1"
react-native: "0.31.0"
react-native-camera: "^0.3.8"

When adding "import Camera from 'react-native-camera';" statement, I get the following error:

> 
> JS Exception: Seems you're trying to access 'ReactNative.createClass' from the 'react-native' package. Perhaps you meant to access 'React.createClass' from the 'react' package instead?
> 
> For example, instead of:
> 
>   import React, { Component, View } from 'react-native';
> 
> You should now do:
> 
>   import React, { Component } from 'react';
>   import { View } from 'react-native';
> 
> Check the release notes on how to upgrade your code - https://github.com/facebook/react-native/releases/tag/v0.25.1
> 

[image]

Removing the import line causes error to go away. Everything was working correctly before latest update to react and react-native.

J-Gonzalez, 2016-08-09:
index.ios.js and index.android.js need to be updated to use upgraded requirerments:
### Old:
```javascript
var React = require('react-native');
var { StyleSheet, requireNativeComponent, PropTypes, NativeModules, DeviceEventEmitter } = React;
```
### Fixed:
```javascript
 var React = require('react');
 var { PropTypes } = React;
 var { StyleSheet, requireNativeComponent, NativeModules, DeviceEventEmitter } = require('react-native');
```

This would most likely be a breaking change for apps using older versions of react-native?

nodegin, 2016-08-11:
awaiting for a fix for this

Oblongmana, 2016-08-18:
^ Short-term fix to avoid the need for making this change any time you `npm install` - change the reference in your package.json to:
`"react-native-camera": "github:oblongmana/react-native-camera#fc20c0e71f50d2376de468623eaea3b0afef2a0e",`

Or better - fork my branch at that SHA, so you have your own copy. That commit is one commit on from the one retrieved when doing a vanilla install from npm, and includes only the fix mentioned above.

realjamie, 2016-09-08:
+1

[Environment]
Node.js version: 4.2.4
React Native version: 0.32.0
React Native platform + platform version: iOS

[react-native-camera]
Version: 0.3.8

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco