react-native-camera/react-native-camera #363: App not working with latest react-native version

tarang9211, 2016-08-10:
### Steps to reproduce
1. git clone https://github.com/spencercarli/react-native-snapchat-clone
2. `rm -rf node_modules` and modify `react-native` version in `package.json`  
3. `npm install`

### Expected behaviour
Should open the camera

### Actual behaviour
This is the error I receive.
[image]

### Environment
- **Node.js version** 6.3.0
- **React Native version**: 0.30.0
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**: git+https://github.com/lwansbrough/react-native-camera.git (latest)

sartrey, 2016-08-24:
upgrade to 0.31.0 & get crash
not sure if react-native camera cause this

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco