react-native-camera/react-native-camera #364: Error setting property 'captureAudio' of RCTCamera with tag #5

EverChris, 2016-08-11:
`<Camera
				ref='camera'
				style={styles.camera}
				aspect={Camera.constants.Aspect.Fill}
				onBarCodeRead={this._onBarCodeRead.bind(this)}>
				<View style={styles.container}>
					<View style={styles.cameraView}>
						<View key="1" style={[styles.borderLeftTop,styles.borderBox]}/>
						<View key="2" style={[styles.borderRightTop,styles.borderBox]}/>
						<View key="3" style={[styles.borderLeftBottom,styles.borderBox]}/>
						<View key="4" style={[styles.borderRightBottom,styles.borderBox]}/>
					</View>
					<Text style={styles.infoText}>
						请将二维码放到框内
					</Text>
				</View>
				{ closeIcon }
			</Camera>`

2016-08-11 10:12:56.923 [info][tid:main][RCTCameraManager.m:266] Error setting property 'captureAudio' of RCTCamera with tag #5: capturing audio

why?

EverChris, 2016-08-11:
and the 'onBarCodeRead' doesn't work, the callback function never be called.

help!!

vinayr, 2016-08-12:
That error seems to be a bug. You won't see that error if you set `captureAudio={false}`. 

For `onBarCodeRead` try something like this -  
```
_onBarCodeRead = (data) => {
  console.log("bar code data", JSON.stringify(data));
}
<Camera onBarCodeRead={this._onBarCodeRead} captureAudio={false}>
.....
```

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco