react-native-camera/react-native-camera #366: Avoid asking for microphone permission

vinayr, 2016-08-11:
I am using this module for QR code scanning. My app doesn't need microphone access. How can I have this module not to ask for microphone access?

This issue has been raised before(#205) and I have tried setting `captureAudio={false}` but still it asks for microphone access.

vinayr, 2016-08-12:
I forked and disabled logic related to `requestAccessForMediaType: AVMediaTypeAudio`

closed 2016-08-12 by vinayr

ataomega, 2016-10-06:
Could you please share details on how to remove microphone access request?

vinayr, 2016-10-06:
@ataomega actually `captureAudio={false}` works. I was using an old fork of this module which did not have `captureAudio`. Now I just use the master branch with `captureAudio={false}` parameter.

ataomega, 2016-10-06:
Thanks for your answer. may i ask where should i put it? in react native file when i'm calling it or should i edit react-native-camera's project files?

Thanks in advance!

vinayr, 2016-10-06:
In your react-native file. Example -
```
<Camera
  captureAudio={false}
  style={styles.camera}
  .....
  .....
/>
```

HarshitMadhav, 2019-02-28:
@vinayr  you said you did something like this 
I forked and disabled logic related to `requestAccessForMediaType: AVMediaTypeAudio `

can you explain a bit?