react-native-camera/react-native-camera #367: Camera does not appear to work with Android 6 (Marshmellow)

cyphire, 2016-08-11:
### Steps to reproduce
1.  Emulator works under android 4
2. Emulator and actual android 6 device does NOT WORK, crashes on the render with <Camera>
3.

### Expected behaviour
Camera should open up in my app.

### Actual behaviour
Camera crashes the application.  Works in Genymotion 4.0 Android emulator, crashes in any 6.0 emulator or LG G4 (running android 6).

### Environment
- **Node.js version**:3.9.3
- **React Native version**: 0.31.0
- **React Native platform + platform version**: Android 6.0

### react-native-camera
**Version**: npm version or "master": Master

Also - IN any version the following occurs but may not be related:

My camera no longer works as well.

React Native 0.31.0
Android 6 does NOT WORK either in the emulator or on an actual device.  The <Camera> crashes the app with no explaination.

Android 4 (Emulator) still DOES work.

This is perplexing.

Also this code snippit says that the module is undefined:

Error: TypeError: undefined is not a function (evaluating '_reactNativeCamera2.default.checkDeviceAuthorizationStatus()')

```
  test_permissions() {
	try {

	  console.log(Camera.checkDeviceAuthorizationStatus());
	} catch(e) {
	  console.log('Error: ' + e);
	}
  }
```

anotherstarburst, 2016-08-15:
Getting the same error:

`java.lang.ExceptionInInitializerError
	at com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation(RCTCameraView.java:112)
	at com.lwansbrough.RCTCamera.RCTCameraView.<init>(RCTCameraView.java:28)
	at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:17)
	at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:7)
	at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:44)
	at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:211)
	at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:148)
	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:890)
	at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:863)
	at com.facebook.react.uimanager.GuardedChoreographerFrameCallback.doFrame(GuardedChoreographerFrameCallback.java:32)
	at com.facebook.react.uimanager.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:131)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:918)
	at android.view.Choreographer.doCallbacks(Choreographer.java:695)
	at android.view.Choreographer.doFrame(Choreographer.java:628)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:906)
	at android.os.Handler.handleCallback(Handler.java:739)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:158)
	at android.app.ActivityThread.main(ActivityThread.java:7229)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.io.Console java.io.Console.printf(java.lang.String, java.lang.Object[])' on a null object reference
	at com.lwansbrough.RCTCamera.RCTCamera.acquireCameraInstance(RCTCamera.java:33)
	at com.lwansbrough.RCTCamera.RCTCamera.<init>(RCTCamera.java:321)
	at com.lwansbrough.RCTCamera.RCTCamera.<clinit>(RCTCamera.java:15)`

timsuchanek, 2016-08-15:
Same here. Android 6.0.1, newest version of this library, react native 0.30.
It seems that the camera is not relased correctly, so when restarting the app, it crashes as it can't connect to the camera.

```sh
08-15 15:26:04.646 23317 23317 E AndroidRuntime: java.lang.ExceptionInInitializerError
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.lwansbrough.RCTCamera.RCTCameraView.setActualDeviceOrientation(RCTCameraView.java:120)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.lwansbrough.RCTCamera.RCTCameraView.<init>(RCTCameraView.java:33)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:18)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.lwansbrough.RCTCamera.RCTCameraViewManager.createViewInstance(RCTCameraViewManager.java:8)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.facebook.react.uimanager.ViewManager.createView(ViewManager.java:44)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:211)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:148)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:890)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:863)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.facebook.react.uimanager.GuardedChoreographerFrameCallback.doFrame(GuardedChoreographerFrameCallback.java:32)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.facebook.react.uimanager.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:125)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:856)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.view.Choreographer.doCallbacks(Choreographer.java:670)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.view.Choreographer.doFrame(Choreographer.java:603)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:844)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:739)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:95)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:148)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:5417)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
08-15 15:26:04.646 23317 23317 E AndroidRuntime: Caused by: java.lang.RuntimeException: Fail to get camera info
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.hardware.Camera._getCameraInfo(Native Method)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at android.hardware.Camera.getCameraInfo(Camera.java:236)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.lwansbrough.RCTCamera.RCTCamera.<init>(RCTCamera.java:313)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        at com.lwansbrough.RCTCamera.RCTCamera.<clinit>(RCTCamera.java:15)
08-15 15:26:04.646 23317 23317 E AndroidRuntime:        ... 22 more
08-15 15:26:04.648   779  1579 W ActivityManager:   Force finishing activity com.wizart/.MainActivity
0

```

cyphire, 2016-08-17:
First of all Loch, thanks for making this module!

I'm building a pretty cool app which needs camera access as it's major part.  Do you know why it is failing or perhaps could point me in the right direction on it? This is integral to my demonstration to investors next week, I don't want to take a step back!

Any help would be appreciated!

charpeni, 2016-08-28:
Hi @cyphire,

Does #381 fixed this issue ?

How it was with investors?

cyphire, 2016-08-28:
Hey Nicolas,  yes thank you, this lets the camera work under 6 now.  I did
have to move my build target to 22. When it was on 23, the camera doesn't
come up.

I just started using react-native-image-picker as an alternative as well.
It gives me quick response time (this module takes a LONG time to focus and
return results).  The problem I have is that I need fast response time and
square camera results.  I was able to achieve this with picker, and it
works.  I would rather, of course, have the camera viewfinder directly in a
view like with rncamera but I've spent too much time figuring out how to
crop, or display the results without stretching or filling the image with
huge white space on either side.

The first investor is starting a huge startup (he is a very, very big guy
in tech and an awesome guy) and his team was afraid my venture would
distract them.  I understand.  I'm going to propose a straight investment,
we'll see!.  Working on some other super high net wealth guys... Wish me
luck!

I haven't written anything relevant in 15 years as I've been running
companies.  5 months ago I taught myself JavaScript through react-native.
I've been a programmer for over 30 years, but again, haven't had time to
code since I had a company in Silicon Alley in NYC during the bubble.  My
JavaScript wasn't just rusty, it sucked.  Now I'm proud to say, my
JavaScript is good and I didn't just build a prototype for investors, I
actually built the whole app and it's a thing of beauty!

Can't wait to get the company rolling and turn this project over to young
guys like you and start building out the company!

_________________________________________
Brian B. Canin    brian.canin@gmail.com

858-945-5601 (cell)

cyphire, 2016-08-28:
Closing this as the new fix allows the camera to work under 6.0 in Android.

Build target needs to target 22, not the latest 23 though:
        targetSdkVersion 22

or the camera does not open, but it no longer crashes my app.

Thanks guys!

closed 2016-08-28 by cyphire