react-native-camera/react-native-camera #369: Captured Image orientation error

RichardBoyewa, 2016-08-16:
### Steps to reproduce
1. I followed the instruction on the page
2. Ran the application
3. Captured the image to disk; select disk as target desination

### Expected behaviour
Image orientation should be kept

### Actual behaviour
Rotates a 90 deg

### Environment
- **Sony Xperia Z2**:
- **React Native version: 0.30**:
- **React Native platform + platform version**: Android 6.0, etc

### react-native-camera
**Version**: npm version or "master"

rasom, 2016-08-26:
#309 the same issue

charpeni, 2016-08-28:
I'll close this issue since it's a duplicate of #309.

closed 2016-08-28 by charpeni

npomfret, 2017-04-26:
@RichardBoyewa  are you able to test my PR?  It may help with this https://github.com/lwansbrough/react-native-camera/pull/682.