react-native-camera/react-native-camera #376: Error when import the Camera from 'react-native-camera';

Talkwondo, 2016-08-22:
When I add it I get this error
[image]

cavinsmith, 2016-08-26:
Identical thing happened to me:

`[error][tid:com.facebook.react.JavaScript] undefined is not an object (evaluating 'CameraManager.CaptureQuality.high')`

benjick, 2016-08-28:
Getting the same error when running on Android (device)

cavinsmith, 2016-08-28:
@Talkwondo, @benjick  I've resolved my problem by uninstalling+unlinking software and installing again via automatic install provided in description. I guess you should try to uninstall, unlink react-camera and try one-by-one all installation methods provided by author.

Panda313, 2016-08-30:
On IOS, "Simulator > Reset Content & Settings" did the trick for me (I did auto and manual linking before but still got the error)

VincentLaulagnet, 2016-08-31:
I have the same issue on Android.
It is working using RN0.28, but stopped working using RN0.29
@cavinsmith  I tried reinstalling and relinking but it did not help.

VincentLaulagnet, 2016-08-31:
The issue in my set-up was related to the linking set-up. I could get it to work by manually modifying the MainApplication.java which was not correctly updated by either react-native link or rnpm link commands

sibelius, 2018-01-07:
this is a problem with react-native link

closed 2018-01-07 by sibelius