react-native-camera/react-native-camera #377: Supply absolute recording URL.

macdabby, 2016-08-23:
Right now, it looks like if you supply Camera.constants.CaptureTarget.disk or Camera.constants.CaptureTarget.temp as the captureTarget, it will create a URL to save the video. It would be great, assuming this would work, if the absolute file path could be generated beforehand and supplied as an argument. This way while the video is recording to disk it can also be read and uploaded to a server.

macdabby, 2016-08-23:
I have a pull request with this feature for iOS and it appears to be working with another thread reading the file while it's recording. Android still needs to be implemented. https://github.com/lwansbrough/react-native-camera/pull/378

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco