react-native-camera/react-native-camera #382: Argument 1 (RCTResponseSenderBlock) of RCTCameraManager.capture must not be null

ChrisPikul510, 2016-08-25:
### Steps to reproduce
`this.camera.capture().then( (data) => console.log(data) )`

### Expected behaviour
It should take the picture

### Actual behaviour
Big red screen. With the error:
`Argument 1 (RCTResponseSenderBlock) of RCTCameraManager.capture must not be null`

Strangely, if I change the code to:
`this.camera.capture({}).then( (data) => console.log(data) )`
the error changes to: 
`Argument 1 ({}) of RCTCameraManager.capture should be a function.

Then changing to:
`this.camera.capture(() => ({})).then( (data) => console.log(data) )`
changes the error to:
`undefined is not an object (evaluating '_this.camera.capture(function(){return{};}).then')

### Environment
- **Node.js version**: 6.4.0
- **React Native version**: 0.30.0
- **React Native platform + platform version**: iOS

### react-native-camera
**Version**: "github:lwansbrough/react-native-camera#1e092b557367fed952554dc025c4cae179651957"

brgarciarivas, 2016-09-01:
Running into the same error , were you able to get a fix?

ChrisPikul510, 2016-09-04:
I believe it actually ended up being something totally different. Seems the debug environment doesn't work as well as intended. A null value in my reducer way down the pipeline somehow produced this error.

realjamie, 2016-09-08:
+1

[Environment]
Node.js version: 4.2.4
React Native version: 0.32.0
React Native platform + platform version: iOS

[react-native-camera]
Version: 0.3.8

prabhaav, 2016-10-14:
Seeing the same issue still, has anyone found a solution?

[Environment]
Node.js version: 6.4.0
React Native version: 0.35.0
React Native platform + platform version: iOS

[react-native-camera]
Version: 0.3.8

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco