react-native-camera/react-native-camera #383: Question : Barcode guide line

Prash88, 2016-08-26:
How to have barcode guidelines/a viewfinder to appear on the camera view. is there a option built in ?

LeBlaaanc, 2016-08-30:
@Prash88 we use the QR code reader and just draw a box with other components, not sure if that's what you were expecting but it works well and might in your case?

M1chaelTran, 2016-12-17:
@Prash88 you can check out how we do it via our fork commit: https://github.com/SkribbleME/react-native-camera/commit/a6e663c93f9023ba86ba19a6894f6d9e3a5528ab

usage along the line of: 
```
<View style={styles.container}>
        <Camera
          onBarCodeRead={this._handleScan}
          captureAudio={false}
          aspect={Camera.constants.Aspect.fill}
          showViewFinder={true}
          viewFinderBorderWidth={5}
          viewFinderShowLoadingIndicator={this.state.showLoadingIndicator}
          style={styles.preview}>
          {this.state.dialog && this._renderDialog()}
        </Camera>
      </View>
```

We also implement 3 seconds delay between scan too

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco