react-native-camera/react-native-camera #384: ReactNative.createClass

benjick, 2016-08-28:
### Steps to reproduce
1. Install camera with npm and rnpm
2. 
```js
import Camera from 'react-native-camera';
<Camera
    style={s.preview}
    aspect={Camera.constants.Aspect.fill} >
  <Text style={s.capture}>[CAPTURE]</Text>
</Camera>
```

### Expected behaviour
To render the camera

### Actual behaviour
It displays this error message:

[image]

### Environment
- **Node.js version**: 6.3
- **React Native version**: 0.32.0
- **React Native platform + platform version**: iOS 9.3

### react-native-camera
**Version**: 0.3.8

benjick, 2016-08-28:
Using `npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save` it works

VincentLaulagnet, 2016-08-31:
I have the same issue on Android. 
It is working using RN0.28, but stopped working using RN0.29
@benjick I tried reinstalling and relinking but it did not help.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco