react-native-camera/react-native-camera #386: ios 10 - app crashes when view with <Camera> is rendered

romanmandryk, 2016-08-29:
react-native-camera crashes the iOS app when built in XCode 8 beta running on ios 10 simulator.
Same code works fine when built with XCode 7 and runs on iOS 9.3.

### Steps to reproduce
1. Install react-native-camera as described in README
2. In Info.plist set NSCameraUsageDescription to any value (required in ios 10)
3. create a sample view with <Camera>

### Expected behaviour
Camera preview should open

### Actual behaviour
After navigating to the view with <Camera>
the app crashes with EXC_BAD_ACCESS. XCode stops in RCTCameraManager.m:34
   self.previewLayer = [AVCaptureVideoPreviewLayer layerWithSession:self.session];

### Environment
- **Node.js version**: 4.4.5
- **React Native version**: 0.31
- **React Native platform + platform version**: iOS 10.0 simulator, XCode 8 beta

### react-native-camera
**Version**: latest  "master" (29 Aug 2016)

### Note
I tried to downgrade to npm version of react-native-camera (0.38) and the app crashes immediately after start (also in RCTCameraManager.m in init method)

Any ideas how to fix this or at least how to debug it?

nitrogenlabs, 2016-09-12:
Running into the same prob.

Environment:

- NodeJS: 6.5.0
- React Native: 0.33
- XCode: 8.0 (8A218a)

derdav3, 2016-09-12:
Had the same, but deleting the build folder, restarting the server with `react-native start -- --reset-cache` solved it

nitrog7, 2016-09-13:
Just to clarify, you have to also use the master branch, not 0.3.8.

ibussieres, 2016-09-13:
I can't get it to work, even with the proposed workaround.

nitrog7, 2016-09-13:
You'll also have to comment out a few lines for the preview on simulator in the initial view. With the master branch, in RCTCameraManager.m (line 30) update the following:

```
- (UIView *)view
{
  self.session = [AVCaptureSession new];
    
#if !(TARGET_IPHONE_SIMULATOR)
  self.previewLayer = [AVCaptureVideoPreviewLayer layerWithSession:self.session];
  self.previewLayer.needsDisplayOnBoundsChange = YES;
#endif
    
  return [[RCTCamera alloc] initWithManager:self bridge:self.bridge];
}
```

Or you can just use a modified fork until the main branch is updated at: https://www.npmjs.com/package/@nitrog7/react-native-camera

`npm install --save @nitrog7/react-native-camera`

dijonkitchen, 2016-09-17:
On the stable release of Xcode 8, I edited the Info.plist for NSCameraUsageDescription as mentioned here: https://developer.apple.com/library/content/documentation/General/Reference/InfoPlistKeyReference/Articles/CocoaKeys.html#//apple_ref/doc/uid/TP40009251-SW24

After just adding the #if and #endif lines to the RCTCameraManager.m file as noted by @nitrog7 and restarting React server, the camera works and doesn't crash on use. 

I didn't have to use master instead of 0.3.8 of react-native-camera.

rgoldiez, 2016-09-17:
Crashing on iPhone 7 w/ iOS 10.  My app was working fine on iPhone 6s w/ iOS 10.

derdav3, 2016-09-17:
error code? @rgoldiez

nitrogenlabs, 2016-09-17:
Thanks, I’m still hoping there is a better solution than to comment out but looking back at my code, I have the NSCameraUsageDescription set and was still getting the error. Not sure if it was because if a beta version of XCode (it shouldn’t be, since I’m on 8A218a) or if there is a cache that has not cleared.

Using the master branch does resolve errors for use with React-Native 0.33 and React 15.3.1.

rgoldiez, 2016-09-17:
@derdav3 - Unfortunately I can't (or don't know how) to run in debug on my device right now as I am at a conference on wifi that blocks traffic to other clients.  I will have to try in debug mode when I get home tonight/tomorrow.

dijonkitchen, 2016-09-18:
@nitrogenlabs it may be because of the beta, try the stable release. 

@rgoldiez if you click on your project name in Xcode, you'll see "Edit scheme" then there's a drop down for release or debug.

rgoldiez, 2016-09-19:
@nitrogenlabs - Please see the screenshot below regarding the crash once the camera loads.  It had been crashing in the simulator, but adding the #if/#endif bypassing where the simulator was crashing, when running in the simulator (the crash in the simulator seems more informative than the screenshot below).

[image]

@dijonkitchen - Thanks ... my issue wasn't know how to run in debug mode... it was that I was at a conference and the wifi did not allow connections to peer clients (ie, running on my device in debut mode and pulling the JS bundle from my laptop).

rgoldiez, 2016-09-19:
Here's the crash in the simulator, when I comment out the #if/#endif that this thread suggested adding to avoid the crash in the simulator.

[image]

dijonkitchen, 2016-09-19:
I didn't actually comment them out with "//'s", I just added those lines 34 and 37 since they weren't there originally. Don't forget to restart React server too. 

When you see the signal SIGABRT error, check your Xcode logs for a detailed response. Don't forget to set the NSCameraUsageDescription in Info.plist as noted above: https://github.com/lwansbrough/react-native-camera/issues/386#issuecomment-247786699

rgoldiez, 2016-09-19:
@dijonkitchen - I added the // comments to show the error in the simulator.  It's crashing at the same point on my iPhone 7 / iOS 10, but with seemingly less useful logs when running on the device.  Maybe unrelated.  

I'll check the Xcode logs for more details.  Also, I already had NSCameraUsageDescription in my info.plist, required now to get a build thru test flight.

rgoldiez, 2016-09-20:
@derdav3 - Any ideas from the screenshots I posted?

ioannagian, 2016-09-20:
@rgoldiez I had the exact same issue and was fixed by adding the two lines as @nitrog7 suggested. They are not comments, they are not to be commented out with "//" !

rgoldiez, 2016-09-20:
@ioannagian --- I **know** they are not comments.  I **only** added the // to the simulator so show the failure in the simulator.  My issue is that the camera component is now crashing **on the actual phone**.  Commenting out this code in the simulator prevents the simulator from crashing.  It still crashes on the phone.

I've been using this component for 6 months without issue.  Only noticed it recently on my iPhone 7.

nitrog7, 2016-09-20:
@rgoldiez Will be able to play around with it on an iPhone 7 later this week. Currently testing on the iPhone 6s. Looking at the change log, I've noticed a lot of small changes to the camera API. May need to update a couple more attributes or defaults

rgoldiez, 2016-09-20:
Thank you @nitrog7 !!

jasongaare, 2016-09-20:
Having the same issue. Running iOS 10 on an iPhone 6. 

My issue is when I build a release on my device, the app works as expected until the application is terminated (double tap home, dismiss app). Then on each subsequent launch the app crashes during launch. When I comment out the lines

`self.previewLayer = [AVCaptureVideoPreviewLayer layerWithSession:self.session];
  self.previewLayer.needsDisplayOnBoundsChange = YES;`

the application launches, but (obviously) the camera doesn't work.

zhipengbird, 2016-09-21:
Add to your Info.plist:

```
<key>NSCameraUsageDescription</key>
<string>Allow us to scan documents and capture images.</string>
```

rgoldiez, 2016-09-21:
@zhipengbird - Already have `NSCameraUsageDescription` in Info.plist.  That's not why it's crashing.

jasongaare, 2016-09-21:
Ends up my error is not iOS 10 specific - I am getting the same crash on iOS 9. Back to the drawing board...

kLkA, 2016-09-22:
Hey guys, any plans when will be release fixing this issue? `NSCameraUsageDescription` in info.plist not helps

dijonkitchen, 2016-09-22:
I've been running it again and could it be the microphone access? In Chrome DevTools, it says microphone is in use when the camera pulls up. 

In Xcode I get this error too: 

> This app has crashed because it attempted to access privacy-sensitive data without a usage description.  The app's Info.plist must contain an NSMicrophoneUsageDescription key with a string value explaining to the user how the app uses this data.

rgoldiez, 2016-09-22:
@ronaldheft - Not sure about a PR but commenting out those lines (or sticking and #if / #endif around them) doesn't fix the problem on an actual device.

rgoldiez, 2016-09-22:
@dijonkitchen - Good find... the NSMicrophoneUsageDescription setting allows the camera to run on my actual device (at least so far, in debug mode)!

Here's the part I don't like.... one of my app is using this camera component **ONLY** for scanning bar codes.  I can see some users being skeptical why barcode functionality requires access to the mic.  Seems to me that this component should only require access to the mic if you are using video mode (or both camera and video) when you launch `<Camera />`.

dijonkitchen, 2016-09-22:
@rgoldiez agreed! Not sure where that is in the source code, but you're welcome to make a pull request that doesn't require that.

rgoldiez, 2016-09-22:
@dijonkitchen - Thanks very kind of you to believe that I understand the Obj C / Swift side of things!  I'll be waiting until someone much smarter than me figures out how to fix this.  

Thank you for the fix for now!

jasongaare, 2016-09-22:
@dijonkitchen @rgoldiez  I've had issues with the camera asking for microphone a lot in using this library with iOS. Whenever I use it I always include `captureAudio={false}` in my props for the `<Camera>` component. Hope this helps. 

(I'm still having this issue even with an NSMicrophoneUsageDescription)

sscaff1, 2016-09-22:
I think this just needs to be added to the Info.plist for the app.

https://developer.apple.com/library/content/documentation/General/Reference/InfoPlistKeyReference/Articles/CocoaKeys.html#//apple_ref/doc/uid/TP40009251-SW1

Scroll down to where it describes NSMicrophoneUsageDescription and it states:

> Important: To protect user privacy, an iOS app linked on or after iOS 10.0, and which accesses any of the device’s microphones, must statically declare the intent to do so. Include the NSMicrophoneUsageDescription key in your app’s Info.plist file and provide a purpose string for this key. If your app attempts to access any of the device’s microphones without a corresponding purpose string, your app exits.

I don't think this is a bug with the repo necessarily.

kLkA, 2016-09-22:
I've added NSMicrophone too and camera started work. Thanks for comment about that :)

bikashsharmabks, 2016-09-23:
Adding NSMicrophoneUsageDescription worked for me on iPhone but on simulator the app crashes 

So as described by @nitrog7 the work around to it for simulator is to have if else block around below block of code in RCTCameraManager.m file

```javascript
#if !(TARGET_IPHONE_SIMULATOR)
  self.previewLayer = [AVCaptureVideoPreviewLayer layerWithSession:self.session];
  self.previewLayer.needsDisplayOnBoundsChange = YES;
#endif
```

Thanks

fud, 2016-09-23:
@bikashsharmabks Did you get issues with "        [self.session commitConfiguration];"  in the RTCameraManager.m after your change?

bikashsharmabks, 2016-09-23:
@fud I didnt get any such issues here is what my setup looks like:

"react": "15.3.1",
"react-native": "^0.33.0",
"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git"

Below changes done for RCTCameraManager.m file
[image]

Thanks

fud, 2016-09-23:
@bikashsharmabks That's exactly what I've done and installed. Mind you I could be doing something else wrong as I'm using the stock standard example mentioned in the readme and have never used this lib before.

romanmandryk, 2016-09-24:
Confirming that after adding "if block" as posted by @nitrog7 , camera works as expected. Thank you.

romanmandryk, 2016-09-24:
I couldn't find PR from @nitrog7 or anyone else, so created one. This fixes my original issue.

nitrog7, 2016-09-24:
I'll add a pull request. Just wanted to make sure it was the best solution and tested.

nitrog7, 2016-09-24:
Here ya go...
https://github.com/lwansbrough/react-native-camera/pull/424

ibussieres, 2016-09-27:
Same problem as @fud for me, on 0.34. Bot only the second time around. iOS asks for access to the camera, I authorize, camera opens. When I come back, it crashes.

rgoldiez, 2016-09-30:
Quick update... it seems that the`NSMicrophoneUsageDescription` is not required in Info.plist if you are only using this component as a camera **and** you `captureAudio={false}` (even tho the docs state that the setting is only applicable to video mode).

rafayhamid2, 2016-10-03:
I have the same issue as @fud described
Did you get any solution for this?

rgoldiez, 2016-10-03:
It's working fine for me with the NSCameraUsageDescription in my info.plist

michaelswe, 2016-10-03:
@rgoldiez Does it also work when you invoke a camera view more than once?
Ref. issue: #426

rgoldiez, 2016-10-03:
I'm not sure. I use <Camera /> in one spot and use it as a barcode/QR reader.

royisch, 2016-10-04:
so what is the solution i dont understand? do i use the master and add the code or it should work? because using 0.3.8 fails to work. there are a lot of threads, could someone write the required solution? 
thanks

ibussieres, 2016-10-04:
If you add the NSCameraUsageDescription, it should work fine on master, except for the second invocation as mentioned earlier. This is another issue, #426, that I made a pull request for a few days ago.

scerelli, 2016-10-06:
People struggling with `SIGBRT` error must add privacy strings in xcode plist.
You have to be sure that you adding microphone permission too and not only the camera once so:

```
Privacy - Microphone Usage Description
Privacy - Camera Usage Description
```

neilsarkar, 2016-10-08:
In case anyone coming here isn't super familiar with xcode or Info.plist, the two comments above are correct. They mean you'll want to open the project in xcode, find `Info.plist` in the file navigator, and add the string values to be displayed to the user in the permissions dialog.

[image]

royisch, 2016-10-09:
Hi @scerelli @ibussieres - im still struggling, sorry. I pulled out the master branch, meaning i installed version 0.3.8 and not 

> npm install --save lwansbrough/react-native-camera

still i get an error. Mostly i think that from all this thread i dont really understand what the issue is and what are the steps to resolve it.

[image]

would be great if someone could write more descriptive steps on what should happen here.

using:
 "react": "^15.3.1"
 "react-dom": "^15.3.1"
 "react-native": "^0.33.0"
 "react-native-camera": "^0.3.8"

jondot, 2016-10-09:
Hey Royi
Here's the summary:

Due to simulator changes as part of iOS 10 updates, the piece of code you bumped into makes it crash. It doesn't happen on device. To avoid crashing use the pragma mentioned above, or take one of the branches that include it (im on mobile so can't maneuver)

Additional issue is Apple having a refreshed permission model which now demands a special entry for each permission. Use the ones from this thread as well. It just happens that an audio permission is needed because this component does video as well, for that if you don't have video, you can disable audio capturing.
If any permission is not treated this way and you still use it - on iOS 10 your app will crash at runtime.

royisch, 2016-10-09:
@jondot - still having issues:
here is the error:
[image]
here is the Plist:
[image]

any suggestions?

wardnath, 2016-10-11:
@neilsarkar that fixed the issue for me, thank you

scerelli, 2016-10-12:
@royisch you have to use `"@nitrog7/react-native-camera": "^0.3.9"` in your `package.json`, import it as `import Camera from '@nitrog7/react-native-camera'` and remember to declare in you plist permissions for microphone and camera

kesha-antonov, 2016-11-12:
[image]

Got this error.

Have all permissions (for camera/microphone/photos)

gabriel, 2016-11-14:
I'm also getting a BAD_ACCESS on `[self.session commitConfiguration]` line in the simulator

abcd-ca, 2017-02-06:
In my case I couldn't reproduce the SIGABRT crash in the sim but could with an iPhone SE running 10.1.1. I added the camera AND the mic permission strings (mentioned above) in Info.plist and that worked around the problem but since I'm only taking still photos, it's suspicious why I'd be prompting the user about the mic – yes, the user sees mention of the microphone in the permission prompt. I also tried putting `captureAudio={false}` on the Camera component in my JSX code but it didn't help. Looking for a real solution now where the mic permission isn't needed – I'll start digging into the source code but let me know if someone's already solved this please.

dijonkitchen, 2017-02-06:
I think all the original issues were solved with the merged PRs previously. 

@abcd-ca I think your problem may be related to #544.

[14 more replies]

closed 2017-05-26 by romanmandryk