react-native-camera/react-native-camera #388: Android video flipped rotated

kLkA, 2016-08-30:
### Steps to reproduce
1. code
```javascript
<Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={styles.preview}
          captureTarget={Camera.constants.CaptureTarget.disk}
          captureMode={Camera.constants.CaptureMode.video}
          captureQuality={Camera.constants.CaptureQuality.medium}
          type={Camera.constants.Type.front}
          orientation={Camera.constants.Orientation.portrait}
          aspect={Camera.constants.Aspect.fill}>

        </Camera>
```
2

### Expected behaviour
This code creates video flipped for 180degrees when watching after saving
Tried to use PortraitUpsideDown - both types and even auto making flipped video

### Actual behaviour

Video should be simply made without any rotation effects

### Environment
- **Node.js version**: 6.2.2
- **React Native version**: 32.0-rc.0
- **React Native platform + platform version**: Android 6.0 / LG Nexus 5

### react-native-camera
"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git",

kLkA, 2016-08-30:
Tested with back camera - works well,
Front camera issue only

kLkA, 2016-08-31:
guess it refers to you @marcejohnson  :)

marcejohnson, 2016-08-31:
Hi @kLkA thanks for the report.  A few things to check:

1. Does this behavior occur when you try the example test program provided here:  https://github.com/lwansbrough/react-native-camera/tree/master/Example
1. Are you seeing this on different Android phone models?   (Folks have seen camera sensor differences per platform).  
1. The community addressed Nexus 5 flipped video here:  https://github.com/lwansbrough/react-native-camera/pull/262#issuecomment-239622268 .  That thread is closed (so no need to post there), but wondering if there something different about your setup or usage?

Thanks again for reporting and investigating.

kLkA, 2016-08-31:
@marcejohnson I checked #262 issue but didnt find this kind of code in latest version of camera while i installed it yesterday. Could you tell me how to check if this code exists in my version of npm module?

marcejohnson, 2016-08-31:
This is the commit from @jiw0220 and @markomatic that addressed Nexus 5 orientation:  

https://github.com/lwansbrough/react-native-camera/pull/262/commits/d2a6209f31dc862a9d4269f4198116c6a05d733a

It was merged into latest.  Does reversing this commit work better for you?

markomatic, 2016-08-31:
Probably nobody tested front camera video recording. As @kLkA noted, this is front camera issue only.

marcejohnson, 2016-08-31:
Interesting.... looks like in commit https://github.com/lwansbrough/react-native-camera/pull/262/commits/d2a6209f31dc862a9d4269f4198116c6a05d733a the following code was removed, which specifically adjusted for front camera.  Any idea why this was requested removed?  Did this cause problems?  

```
int mediaRecorderHintOrientation = (90 + ((720 -    RCTCamera.getInstance().getActualDeviceOrientation() * 90))) % 360;
// adjust for differences in how devices display front facing http://www.theverge.com/2015/11/9/9696774/google-nexus-5x-upside-down-camera		
if (RCT_CAMERA_TYPE_FRONT == options.getInt("type")) {		
    if ((RCT_CAMERA_ORIENTATION_PORTRAIT == RCTCamera.getInstance().getOrientation()) && (RCTCamera.getInstance().getAdjustedDeviceOrientation() == 90)) {		
         mediaRecorderHintOrientation += 180;		
    }		
}
```

@kLkA if you add the above code back into RCTCameraModule.java, is your issue resolved?  You can see in https://github.com/lwansbrough/react-native-camera/pull/262/commits/d2a6209f31dc862a9d4269f4198116c6a05d733a where to place it.

markomatic, 2016-08-31:
For me issue is present even with old code. @kLkA, did you tested it? Reason for removal is in comment https://github.com/lwansbrough/react-native-camera/pull/262#issuecomment-239622268

markomatic, 2016-08-31:
As noted here https://developer.android.com/reference/android/media/MediaRecorder.html#setOrientationHint(int) this method will not trigger the source video frame to rotate during video recording! That means that video will not be properly rotated on some players. Is there any way to actually rotate camera? Should we use some code from image capture, as I see, a lot of probably usable code from image capturing is eliminated in video recording with return statement https://github.com/lwansbrough/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L423

kLkA, 2016-09-01:
Hey, added code to the java file and video captured with proper orientation.
Whats the plan so ? integrate it back or wait for new solution?

markomatic, 2016-09-01:
Integrate it back for now if it works for you.

marcejohnson, 2016-09-01:
hey @kLkA check out how @dojiboy9 addressed rotation on front camera here:  https://github.com/marcejohnson/react-native-camera/pull/7

Does this work for you or is this a similar to the approach you took?  Thanks @markomatic, @kLkA, @dojiboy9 for testing across devices.

dojiboy9, 2016-09-01:
@marcejohnson happy to help. 

@klkA Regarding my pull request, I think this commit is the relevant one: https://github.com/marcejohnson/react-native-camera/commit/1455e974fef37de8211949193fa3c32c376c31fc

kLkA, 2016-09-01:
Hey everyone. Right. commit from @dojiboy9 solving this issue very well. 
Thank you all guys!

marcejohnson, 2016-09-02:
Awesome!
Thanks @corbt for creating a pull request containing @dojiboy9's fix already:  https://github.com/lwansbrough/react-native-camera/pull/393

corbt, 2016-09-02:
Whoops, didn't realize that @dojiboy9 had already looked into this! Better tracking of the issues could have saved me several hours of debugging today. :P

Looks like that PR includes some new handling for `onPause` as well, which should also get merged.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco