react-native-camera/react-native-camera #389: Android onFocusChanged

lazarte, 2016-09-01:
Below code works on IOS but not on android?

<Camera
    onFocusChanged={this.onFocusChanged.bind(this)}
    defaultOnFocusComponent={true}
    ref = {(cam) => {this.camera = cam}}
    style = {styles.preview}
    captureAudio = {false}
    type = {Camera.constants.Type.back}>
</Camera>

unipheye, 2016-09-29:
I sort of have the same issue. Check #432 Auto-focus doesn't work on back camera

Cygnus2112, 2016-11-29:
I am having the same issue as @lazarte. I can't seem to get onZoomChanged to work on android either.

abrahambotros, 2017-01-30:
Hi all, please check out https://github.com/lwansbrough/react-native-camera/pull/575 for tap-to-focus and improved (continuous) auto-focus. I've also noted in the README that indeed `onFocusChanged` and `onZoomChanged` don't actually get called on Android yet, though pinch-to-zoom seems to work (without any of my PR changes) and https://github.com/lwansbrough/react-native-camera/pull/575 should enable tap-to-focus and better auto-focus as you guys have been mentioning.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco