react-native-camera/react-native-camera #395: camera.capture() promise never called

tslater, 2016-09-04:
### Steps to reproduce
1. Clone the example from react-native-camera
2. npm i && react-native run-android
3. Press the camera buttons

### Expected behaviour
See something in the log:
```
  takePicture() {
    if (this.camera) {
      this.camera.capture()
        .then((data) => console.log(data))
        .catch(err => console.error(err));
    }
  }
```

### Actual behaviour
Nothing shows in the log (i.e. promise is never called). I also tried adding a `finally` with no success.

### Environment
- **Node.js version**: 6.5.0 (upgraded after seeing there error, started on 4.4.1)
- **React Native version**: ^0.26.2 (from example)
- **React Native platform + platform version**: Android 6.0.1

### react-native-camera
**Version**: master

tslater, 2016-09-10:
It works if I set the orientation: `orientation={Camera.constants.Orientation.portrait}`, but not if it is the default, which I believe is auto.

editkid, 2017-03-06:
Confirming that this is happening also on some of my Android devices (e.g. Nexus 7) and that setting the orientation does 'fix' the issue.

Satya5614, 2017-03-12:
Setting `orientation={Camera.constants.Orientation.portrait` doesn't fix the problem when the capture mode is Video. Is there any other way to fix this?

editkid, 2017-03-12:
I'm using video capture mode, @Satya5614. For me the workaround works.

Satya5614, 2017-03-12:
@editkid Have you tested that on iOS device?

editkid, 2017-03-12:
I have no problems on iOS at all. The only platform that required the 'fix' for me was Android, and then only on some devices.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco