react-native-camera/react-native-camera #397: RN Camera seems to disable Vibration.vibrate()

cpsiaki, 2016-09-06:
### Steps to reproduce
1.I am using react-native-camera to scan barcodes. The issue is that after the camera renders it seems to disable the barcode

```

      <Camera onBarCodeRead={this._onBarCodeRead} style={styles.camera}>
		<View style={styles.instructions}>
			<Text style={styles.instructionsText}>{'Please place the QR code in the box'}</Text>
		</View>
        <View style={styles.rectangleContainer}>
          <View style={styles.rectangle}/>
        </View>
      <View>
        <TouchableOpacity style={styles.cancelButton} onPress={this._onPressCancel}>
          <Text style={styles.cancelButtonText}>Cancel</Text>
        </TouchableOpacity>
      </View>
      </Camera>
```

my function _onBarCodeRead only has Vibration.vibrate() inside of it. It does not execute.

If I put a Vibration.vibrate() in ComponentWillMount() it works just fine. and if I put a cancel button with a navigator pop and vibrate it will work if I click cancel before the camera becomes fully functional but not after.

Anybody else experience anything like this?

### Expected behaviour
Vibration.vibrate() should shake the iPhone

### Actual behaviour
Vibration.vibrate() does not vibrate

### react-native-camera
iOS (most recent)
0.3.8
RN 0.32

christopherdro, 2016-09-19:
I am able to reproduce this issue. Will try and take a deeper look when I can.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

frangeris, 2020-09-02:
@ospfranco Can you reopen? this is still happening...