react-native-camera/react-native-camera #403: App crashed / App reload when changing the permission of camera

keung725, 2016-09-12:
### Steps to reproduce
App reload:
1. Go to setting, close camera permission, run app
2. Go to camera component
3. Go to setting, open permission
4. App reload

App crashed:
1. Go to camera component
2. Go to setting, close permission
3. App crashed

I don't want App reload and App crashed any more.
How can I handle it with not modifying the objective-c, because I don't know it. Thanks.

ryankask, 2016-09-13:
I think this is expected behaviour as iOS kills apps when privacy permissions change.

smartt, 2016-09-15:
FWIW, I'm running into an app crash as well, but it seems to just be on iOS 10.  I added https://github.com/yonahforst/react-native-permissions to help detect the permissions; but at this point, even with permission granted, it's still crashing even if I just import `react-native-camera`.

pranny, 2016-09-15:
Starting iOS 10, the key `NSCameraUsageDescription` is mandatory in `Info.plist`. Quoting from the [apple developer documentation](https://developer.apple.com/library/content/documentation/General/Reference/InfoPlistKeyReference/Articles/CocoaKeys.html)

>To protect user privacy, an iOS app linked on or after iOS 10.0, and which accesses the device’s camera, must statically declare the intent to do so. Include the NSCameraUsageDescription key in your app’s Info.plist file and provide a purpose string for this key. If your app attempts to access the device’s camera without a corresponding purpose string, your app exits.

I also faced app crash issues and updating `Info.plist` resolved it. Give it a try.

smartt, 2016-09-15:
An update on the crash we're having:

We do have `NSCameraUsageDescription` defined (that was caught on a Test Flight upload) as well as `NSPhotoLibraryUsageDescription`.

When running in Test-mode, XCode's debugger halts in `RCTCameraManager.m`, line 33, with a `EXC_BAD_ACCESS (code=1, address=0x50)`.  Currently, line 33 reads:

```
self.previewLayer = [AVCaptureVideoPreviewLayer layerWithSession:self.session];
```

Any ideas?

rgoldiez, 2016-09-17:
Wasn't happening to me on iPhone 6s w/ iOS (actual device) ....... but iPhone 7 w/ iOS 10 is crashing now in the same way as described earlier in this thread.

krzkaczor, 2016-09-19:
Here's a full bug report and more solutions: https://github.com/lwansbrough/react-native-camera/issues/386

xavicolomer, 2016-09-22:
For me it crashed until I set all Pricacy Settings:

- NSCameraUsageDescription

comountainclimber, 2016-09-26:
experiencing the same issue... :(

JimmyStavros, 2016-11-16:
@smartt hey not sure if you got this working but try nitrog7's solution in #386. Worked like a charm for me.

KarthickHero, 2017-10-12:
I just added exit(0) when i opened the camera settings. It's working fine

victorbadila, 2017-11-03:
@KarthickHero where did you add that one line and for what purpose? Could you give more details?

The issue happens for me on iOS11 on both iPhone 7 and iPhone SE, however it seems to be a general permissions problem, happens when I change permissions for anything, I don't think it is a react-native-camera specific problem.

MIFind, 2017-12-21:
Hey , I have a problem.
Do i still need `react-native-permissions` for private permission , the time when I have seted
`NSCameraUsageDescription
NSMicrophoneUsageDescription`.

mchandleraz, 2018-01-02:
@MIFind I don't quite understand your question, but I recently implemented permissions checking in my app using the `checkVideoAuthorizationStatus` and `checkDeviceAuthorizationStatus`. These methods return a Boolean indicating whether you have the permissions. Check README.md for more info.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco