react-native-camera/react-native-camera #404: Error in RCTCamera:340 HashMap<>()

jacobSingh, 2016-09-13:
### Steps to reproduce
1. react-native android-run
2.
3.

### Expected behaviour
Should start the application

### Actual behaviour

```

/Users/jacob/stuff/CB/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCamera.java:340: illegal start of type
        _cameras = new HashMap<>();
                               ^
/Users/jacob/stuff/CB/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCamera.java:341: illegal start of type
        _cameraInfos = new HashMap<>();
                                   ^
/Users/jacob/stuff/CB/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCamera.java:342: illegal start of type
        _cameraTypeToIndex = new HashMap<>();
                                         ^
3 errors
:react-native-camera:compileReleaseJavaWithJavac FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:compileReleaseJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

```

### Environment
- **Node.js version**: 6.5.0
- **React Native version**: react-native-cli: 1.0.0 react-native: 0.33.0
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
Android 6.0

### react-native-camera
**Version**: npm version or "master": react-native-camera@0.3.8

bogobogo, 2016-09-21:
I am having the same issue. 

Did you manage do solve it? if so how?

jacobSingh, 2016-09-21:
Yes! I upgraded my java version (got the version from Oracle I think which is 1.8). 
java -version
java version "1.8.0_101"
Java(TM) SE Runtime Environment (build 1.8.0_101-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.101-b13, mixed mode)

Then I added this to my zshrc: 
export JAVA_HOME=`/usr/libexec/java_home -v 1.8`

I think I may have also messed with something in Android studio, but try that.

ptomasroos, 2017-02-14:
As seen in the readme.

```
JDK >= 1.7 (if you run on 1.6 you will get an error on "_cameras = new HashMap<>();")
```

closed 2017-02-14 by ptomasroos