react-native-camera/react-native-camera #408: how to interpret returned object

adammenges, 2016-09-13:
### Steps to reproduce
1. capture image
2. log output

```javascript
var x = this.camera.capture().then((data) => console.log(data)).catch(err => console.error(err));
console.log(x);
```

### Expected behaviour
Base64 encoded string according to readme? Maybe?

"By default the image binary is sent back as a base 64 encoded string." -- readme

### Actual behaviour
Object, which I see at the top of the readme is a braking change, but then I have no idea how interpret this thing:

```javascript
{ _45: 0, _81: 0, _65: null, _54: null }
```

### Environment
- **Node.js version**: 5.6
- **React Native version**: 0.32
- **React Native platform + platform version**: iOS 9.0

### react-native-camera
**Version**: "react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git",

adammenges, 2016-09-13:
Also will there be no way to return the base64 images in this new version?

davidperrenoud, 2016-09-27:
The variable `x` will be a Promise, so you cannot use it directly like in your example. You can read more about Promises here:
http://www.html5rocks.com/en/tutorials/es6/promises/

Instead, the variable `data` will receive base-64 encoded string, once the Objective-C/Java code has done its work:
```javascript
this.camera.capture()
.then((data) => {
  console.log(data)
}).catch((err) => {
  console.error(err)
});
```

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco