react-native-camera/react-native-camera #409: Build fails on Android

jrm2k6, 2016-09-13:
### Steps to reproduce
1. npm install react-native-camera@https://github.com/lwansbrough/react-native-camera.git --save + react-native link react-native-camera
2. Create Camera Component
3. react-native run-android

### Expected behaviour
I should see a camera preview

### Actual behaviour
Build fails with 
```
/Users/jrm2k6/Workspace/pocketderm.dev/curology-companion/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java:52: error: cannot find symbol
    implements MediaRecorder.OnInfoListener, LifecycleEventListener {
                                             ^
  symbol: class LifecycleEventListener
/Users/jrm2k6/Workspace/pocketderm.dev/curology-companion/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java:97: error: incompatible types: RCTCameraModule cannot be converted to LifecycleEventListener
        _reactContext.addLifecycleEventListener(this);
                                                ^
/Users/jrm2k6/Workspace/pocketderm.dev/curology-companion/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java:684: error: method does not override or implement a method from a supertype
    @Override
    ^
/Users/jrm2k6/Workspace/pocketderm.dev/curology-companion/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java:689: error: method does not override or implement a method from a supertype
    @Override
    ^
/Users/jrm2k6/Workspace/pocketderm.dev/curology-companion/node_modules/react-native-camera/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java:697: error: method does not override or implement a method from a supertype
    @Override
    ^
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some messages have been simplified; recompile with -Xdiags:verbose to get full output
5 errors
:react-native-camera:compileReleaseJavaWithJavac FAILED

FAILURE: Build failed with an exception.
```

### Environment
- **Node.js version**: v6.2.0
- **React Native version**: react-native: 0.32.1
- **React Native platform + platform version**: Android 5.1

### react-native-camera
**Version**:  "0.3.8"

comountainclimber, 2016-09-13:
looks like the build now fails with react-native .31 as well...

comountainclimber, 2016-09-13:
pointing to any of the commits made prior to https://github.com/lwansbrough/react-native-camera/commit/b01c64e7734827da3f4c61e5f7225f70b454c945 fixes this issue... 

In my package JSON I specify a commit to fallback to before this breaking change was merged into Master: `"react-native-camera": "git+https://github.com/lwansbrough/react-native-camera.git#f4bf30e17b78c89d834625514929956af9d0336f"`

gaguirre, 2016-09-13:
@comountainclimber are you getting the same error than @jrm2k6?
Despite the PR you mention could have introduced an error, I think this issue is not related to that.

Regarding your error, are you getting it from the Example right?

comountainclimber, 2016-09-13:
I am getting the same error in my personal project after running `rm -rf node_modules && npm i` and then `react-native run-android`

jrm2k6, 2016-09-13:
But ios works just fine! I haven't tried any of the previous commits.

gaguirre, 2016-09-13:
Guys, apparently it's a merge issue. In our fork the `LifecycleEventListener` even exists. See [here](https://github.com/underscopeio/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java).

gaguirre, 2016-09-13:
Confirmed, some dependencies were pruned in my commit. One of them was `com.facebook.react.bridge.*` that's used by [this PR](https://github.com/lwansbrough/react-native-camera/commit/046970fecaa0f3bb45c9b0d4ca0dadd8dfa3f556).

I will send a PR with the fix!

Import things using `.*` is not a good practice :(

comountainclimber, 2016-09-13:
thanks @gaguirre !

PR #411 (2016-09-13): Add missing dependency
When merging the #399 a wildcard dependency was pruned
```java
import com.facebook.react.bridge.*
```
One of these classes (`LifecycleEventListener`) was used by another PR (#394).

This fixes #409
closed 2016-09-13 by rt2zz