react-native-camera/react-native-camera #415: Error When try to run Mocha tests - TypeError: Cannot read property 'Aspect' of undefined

ShvedDmutro, 2016-09-19:
**/Users/shved/Projects/mobileapp-javascript-react-defects_manager/node_modules/react-native-camera/index.js:250
}}]);return Camera;}(_react.Component);Camera.constants={Aspect:CameraManager.Aspect,BarCodeType:CameraManager.BarCodeType,Type:CameraManager.Type,CaptureMode:CameraManager.CaptureMode,CaptureTarget:CameraManager.CaptureTarget,CaptureQuality:CameraManager.CaptureQuality,Orientation:CameraManager.Orientation,FlashMode:CameraManager.FlashMode,TorchMode:CameraManager.TorchMode};Camera.propTypes=_extends({},_reactNative.View.propTypes,{aspect:_react.PropTypes.oneOfType([_react.PropTypes.string,_react.PropTypes.number]),captureAudio:_react.PropTypes.bool,captureMode:_react.PropTypes.oneOfType([_react.PropTypes.string,_react.PropTypes.number]),captureQuality:_react.PropTypes.oneOfType([_react.PropTypes.string,_react.PropTypes.number]),captureTarget:_react.PropTypes.oneOfType([_react.PropTypes.string,_react.PropTypes.number]),defaultOnFocusComponent:_react.PropTypes.bool,flashMode:_react.PropTypes.oneOfTy

TypeError: Cannot read property 'Aspect' of undefined
    at Object.<anonymous> (/Users/shved/Projects/mobileapp-javascript-react-defects_manager/node_modules/react-native-camera/index.js:58:13)
    at Module._compile (module.js:541:32)
**

michaelswe, 2016-09-25:
Mocha tests are run by Node, afaik. Node doesn't know anything about native components, i.e. you can't test react-native-camera as it's a native component. You'll have to mock the component.

michaelswe, 2016-11-08:
Did it work out for you with the mocking? If so, we can close this issue. Seems like the maintainer(s) have quite some work to do with open issues. We could at least close the non-issues for them.

yuanfang-fn, 2017-01-13:
how to mock the component.

sibelius, 2017-12-09:
use react-native-mock with mocha to fix this

or use jest instead

closed 2017-12-09 by sibelius