react-native-camera/react-native-camera #416: Width and height on capture for android

mahanhaz, 2016-09-20:
i'm not sure if anyone is working on it or even if it's possible.
i'm trying to reach this goal on android:
```
    this.camera.capture([
      width: windowWidth,
      height: windowWidth,
    ])
      .then((data) => {
        console.log('capture finished', data)
      })
      .catch(err => {
        console.error('error', err)
      });
```
i'm trying to capture a video as we are expecting from the code with specific width and height.
i didn't check the source code yet, i wanted to know if anyone working on it or if not, is it possible to do it, which in this case i would like to make PR for it to fix this issue for everybody.

sundayhd, 2017-01-09:
Its will be awesome :-)

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco