react-native-camera/react-native-camera #417: [Crash] No active/enabled connections

bsiddiqui, 2016-09-20:
Received a crash 
```
Fatal Exception: NSInvalidArgumentException
*** -[AVCaptureMovieFileOutput startRecordingToOutputFileURL:recordingDelegate:] - no active/enabled connections.
```

Camera is using captureSource temp, recording a video, orientation auto

### Environment
- **React Native version**: 0.31.0
- **React Native platform + platform version**: iOS 9.3
- **Device**: iPhone 5s

### react-native-camera
**Version**: master

### Full crash report
```
#22. Crashed: com.twitter.crashlytics.ios.exception
0  Ramble                         0x100136aa8 CLSProcessRecordAllThreads + 4296616616
1  Ramble                         0x100136aa8 CLSProcessRecordAllThreads + 4296616616
2  Ramble                         0x100136964 CLSProcessRecordAllThreads + 4296616292
3  Ramble                         0x100127124 CLSHandler + 4296552740
4  Ramble                         0x100134b7c __CLSExceptionRecord_block_invoke + 4296608636
5  libdispatch.dylib              0x1813b547c _dispatch_client_callout + 16
6  libdispatch.dylib              0x1813c0728 _dispatch_barrier_sync_f_invoke + 100
7  Ramble                         0x100134620 CLSExceptionRecord + 4296607264
8  Ramble                         0x100134454 CLSExceptionRecordNSException + 4296606804
9  Ramble                         0x100134074 CLSTerminateHandler() + 4296605812
10 libc++abi.dylib                0x180fc2f44 std::__terminate(void (*)()) + 16
11 libc++abi.dylib                0x180fc285c __cxxabiv1::exception_cleanup_func(_Unwind_Reason_Code, _Unwind_Exception*) + 134
12 libobjc.A.dylib                0x180fd0094 _objc_exception_destructor(void*) + 330
13 AVFoundation                   0x18809fecc -[AVCaptureMovieFileOutput stopRecording] + 490
14 Ramble                         0x1001f94f0 __68-[RCTCameraManager captureVideo:options:orientation:resolve:reject:]_block_invoke + 4297413872
15 libdispatch.dylib              0x1813b54bc _dispatch_call_block_and_release + 24
16 libdispatch.dylib              0x1813b547c _dispatch_client_callout + 16
17 libdispatch.dylib              0x1813c14c0 _dispatch_queue_drain + 864
18 libdispatch.dylib              0x1813b8f80 _dispatch_queue_invoke + 464
19 libdispatch.dylib              0x1813c3390 _dispatch_root_queue_drain + 728
20 libdispatch.dylib              0x1813c30b0 _dispatch_worker_thread3 + 112
21 libsystem_pthread.dylib        0x1815cd470 _pthread_wqthread + 1092
22 libsystem_pthread.dylib        0x1815cd020 start_wqthread + 4

--

Fatal Exception: NSInvalidArgumentException
0  CoreFoundation                 0x18196ae38 __exceptionPreprocess
1  libobjc.A.dylib                0x180fcff80 objc_exception_throw
2  AVFoundation                   0x18809fecc -[AVCaptureMovieFileOutput stopRecording]
3  Ramble                         0x1001f94f0 __68-[RCTCameraManager captureVideo:options:orientation:resolve:reject:]_block_invoke
4  libdispatch.dylib              0x1813b54bc _dispatch_call_block_and_release
5  libdispatch.dylib              0x1813b547c _dispatch_client_callout
6  libdispatch.dylib              0x1813c14c0 _dispatch_queue_drain
7  libdispatch.dylib              0x1813b8f80 _dispatch_queue_invoke
8  libdispatch.dylib              0x1813c3390 _dispatch_root_queue_drain
9  libdispatch.dylib              0x1813c30b0 _dispatch_worker_thread3
10 libsystem_pthread.dylib        0x1815cd470 _pthread_wqthread
11 libsystem_pthread.dylib        0x1815cd020 start_wqthread

#0. com.apple.main-thread
0  libsystem_kernel.dylib         0x1814e8fd8 mach_msg_trap + 8
1  libsystem_kernel.dylib         0x1814e8e54 mach_msg + 72
2  MediaToolbox                   0x1878f2544 FigPlayerRemoteClient_SetPlaybackItemPropertyFixedSize + 260
3  MediaToolbox                   0x1878f235c itemremote_SetProperty + 1768
4  MediaToolbox                   0x1878f7120 itemremote_makeReadyForInspection + 72
5  MediaToolbox                   0x1879fc64c playerremote_updateLayerDurationsWhenItemReady + 168
6  MediaToolbox                   0x1878ff054 itemremote_postNotificationWithPayload + 2576
7  MediaToolbox                   0x1879fa2b4 playerremote_fetchNotifications + 996
8  MediaToolbox                   0x1878fa354 FigPlayerRemoteCallbacksServer_SendNotifyPing + 132
9  MediaToolbox                   0x1878fa2a4 _XSendNotifyPing + 56
[... 281 lines cut ...]

#23. Thread
0  libsystem_pthread.dylib        0x1815cd01c start_wqthread + 14

#24. Thread
0  libsystem_kernel.dylib         0x181504b48 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x1815cd530 _pthread_wqthread + 1284
2  libsystem_pthread.dylib        0x1815cd020 start_wqthread + 4

#25. Thread
0  libsystem_kernel.dylib         0x181504b48 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x1815cd530 _pthread_wqthread + 1284
2  libsystem_pthread.dylib        0x1815cd020 start_wqthread + 4

#26. Thread
0  libsystem_kernel.dylib         0x181504b48 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x1815cd530 _pthread_wqthread + 1284
2  libsystem_pthread.dylib        0x1815cd020 start_wqthread + 4

#27. Thread
0  libsystem_kernel.dylib         0x181504b48 __workq_kernreturn + 8
1  libsystem_pthread.dylib        0x1815cd530 _pthread_wqthread + 1284
2  libsystem_pthread.dylib        0x1815cd020 start_wqthread + 4

```

dkaufhold, 2017-10-28:
Got this error too.

```
NSInvalidArgumentException: Application threw exception NSInvalidArgumentException:
*** - [AVCaptureStillImageOutput captureStillImageAsynchronouslyFromConnection:completionHandler:] 
Inactive/invalid connection passed
Originated at or in a subcall of RCTFBQuickPerformanceLoggerConfigureHooks
```

OS: iOS 11.0.3
Phone: iPhone 7

The raw traceback I got:

```
  CoreFoundation      0x181333d38  __exceptionPreprocess
  libobjc.A           0x180848528  objc_exception_throw
  at fn(node_modules/react-native-sentry/lib/NativeClient.js:155:29)
  at tryCallTwo(node_modules/promise/setimmediate/core.js:45:5)
  at doResolve(node_modules/promise/setimmediate/core.js:200:13)
  at Promise(node_modules/promise/setimmediate/core.js:66:3)
  at capture(node_modules/react-native-camera/index.js:236:26)
  at onPress(app/containers/CameraScreen.js:95:10)
  at touchableHandlePress(node_modules/react-native/Libraries/Components/Touchable/TouchableOpacity.js:126:38)
  at _performSideEffectsForTransition(node_modules/react-native/Libraries/Components/Touchable/Touchable.js:746:14)
  at _receiveSignal(node_modules/react-native/Libraries/Components/Touchable/Touchable.js:664:12)
  at apply(node_modules/react-native/Libraries/Components/Touchable/Touchable.js:433:10)
  at getStackAddendumByWorkInProgressFiber(node_modules/react-native/Libraries/Renderer/src/renderers/shared/utils/ReactErrorUtils.js:22:10)
  at invokeGuardedCallbackAndCatchFirstError(node_modules/react-native/Libraries/Renderer/src/renderers/shared/utils/ReactErrorUtils.js:133:57)
  at executeDispatch(node_modules/react-native/Libraries/Renderer/src/renderers/shared/shared/event/EventPluginUtils.js:89:19)
  at executeDispatchesInOrder(node_modules/react-native/Libraries/Renderer/src/renderers/shared/shared/event/EventPluginUtils.js:121:5)
  at forEach(node_modules/react-native/Libraries/Renderer/src/renderers/shared/shared/event/EventPluginHub.js:37:22)
  at forEachAccumulated(node_modules/react-native/Libraries/Renderer/src/renderers/shared/utils/forEachAccumulated.js:30:9)
  at processEventQueue(node_modules/react-native/Libraries/Renderer/src/renderers/shared/shared/event/EventPluginHub.js:232:7)
  at runEventQueueInBatch(node_modules/react-native/Libraries/Renderer/src/renderers/shared/shared/event/ReactEventEmitterMixin.js:18:18)
  at call(node_modules/react-native/Libraries/Renderer/src/renderers/shared/shared/event/ReactEventEmitterMixin.js:38:5)
  at batchedUpdates(node_modules/react-native/Libraries/Renderer/src/renderers/shared/stack/reconciler/Transaction.js:150:20)
  at batchedUpdatesWithControlledComponents(node_modules/react-native/Libraries/Renderer/src/renderers/shared/shared/event/ReactGenericBatching.js:51:12)
  at _receiveRootNodeIDEvent(node_modules/react-native/Libraries/Renderer/src/renderers/native/ReactNativeEventEmitter.js:105:26)
  at apply(node_modules/react-native/Libraries/Renderer/src/renderers/native/ReactNativeEventEmitter.js:192:31)
  at fn(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:250:42)
  at __guard(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:218:7)
  at value(node_modules/react-native/Libraries/BatchedBridge/MessageQueue.js:100:10)
  AVFoundation        0x186c0b3e4  -[AVCaptureStillImageOutput captureStillImageAsynchronouslyFromConnection:completionHandler:]
  litboxedNative      0x1003009b0  RCTFBQuickPerformanceLoggerConfigureHooks
  libdispatch         0x180cb9088  _dispatch_call_block_and_release
  libdispatch         0x180cb9048  _dispatch_client_callout
  libdispatch         0x180cc2e48  _dispatch_queue_serial_drain$VARIANT$mp
  libdispatch         0x180cc37d8  _dispatch_queue_invoke$VARIANT$mp
  libdispatch         0x180cc4200  _dispatch_root_queue_drain_deferred_wlh$VARIANT$mp
  libdispatch         0x180ccc4a0  _dispatch_workloop_worker_thread$VARIANT$mp
  libsystem_pthread   0x180f5efe0  _pthread_wqthread
  libsystem_pthread   0x180f5ec30  start_wqthread
```

Happened while calling `this.camera.capture()` in the following component (slightly simplified version):

```
import React, { Component } from 'react'
import PropTypes from 'prop-types'
import {
  TouchableOpacity,
  View,
} from 'react-native'
import { connect } from 'react-redux'
import Camera from 'react-native-camera'

import * as cameraActions from '../actions/camera'

class CameraScreenBase extends Component {
[... 109 lines cut ...]
    "react-addons-test-utils": "15.5.1",
    "react-test-renderer": "15.5.4",
    "validate.js": "0.11.1"
  },
  "jest": {
    "preset": "react-native",
    "verbose": true,
    "bail": true,
    "globals": {
      "NODE_ENV": "test"
    },
    "transformIgnorePatterns": [
      "node_modules/(?!(react-native|react-navigation|react-native-elements|react-native-vector-icons|react-native-keyboard-aware-scroll-view|react-get-form-data|react-native-swiper|react-native-firebase|react-native-sentry|react-native-side-menu|react-native-tab-navigator|react-native-push-notification)/)"
    ],
    "setupFiles": [
      "<rootDir>/__mocks__/setupTestEnv.js"
    ]
  },
  "rnpm": {
    "assets": [
      "./app/assets/fonts/"
    ]
  }
}
```

sibelius, 2017-12-09:
can somebody create a repo to reproduce this?

tommynordli, 2017-12-11:
I'm getting the same error when trying to take a still photo:
Steps to reproduce (I'm using react-navigation for navigation):
1. Navigate to camera view and rendering <Camera>
2. Take photo (this works)
3. Navigate back to home using react-navigation, and resetting the navigation stack
4. Navigate to camera view again, and <Camera> is only rendering a black view
5. Try to take photo now, and I get the same error
6. If I navigate back without trying to take a photo, and then navigating to the camera view again, the camera is working as intended, and is taking photos. 

This doesn't happen when I'm not resetting the navigation stack, but I need to do that. Looks liks the component is having some problems re-mounting?

gurmukhp, 2018-01-14:
I also get the same error, works fine on simulator but on iPad Air 2 and iPhone X on 11.2 I get the same error about an invalid connection.

Has anyone got it to work? Even the example app doesn't build at all (but that's prob a different issue).

SimonErm, 2018-07-24:
related to VideoQuality problem from https://stackoverflow.com/questions/22110911/avcapturemoviefileoutput-no-active-enabled-connections .
You can workaround by setting videoQuality option on record() . I think a default value for videoQuality should fix this, if anyone could do a PR...

SimonErm, 2018-07-24:
related to VideoQuality problem from https://stackoverflow.com/questions/22110911/avcapturemoviefileoutput-no-active-enabled-connections .
You can workaround by setting videoQuality option on record() to 1080p because default is 2160p and don't work. I think updating the default value for videoQuality should fix this, if anyone could do a PR....

thatjuan, 2018-07-25:
@SimonErm is right.. This happens when a `quality` option is not passed to `recordAsync` on iOS. 

The fix should be to set the default option to the maximum supported by the device. If everybody agrees with this solution, I can send in a PR in the next few days.

In the meantime, if you're running into this, try:

```
            let options = {
                quality: RNCamera.Constants.VideoQuality["1080p"]
            };

            this.camera.recordAsync(options)
```

This might fail on devices that don't support 1080p, though.

benjaminreid, 2018-07-30:
I’ve run into this today too. While setting the quality does work around this issue, there’s an issue where setting the quality causes the camera’s zoom to jump...

n1ru4l, 2018-07-30:
@benjaminreid I can also confirm this behaviour

shalin87, 2018-08-09:
Video does not start again. 

```
takePicture = async function () {
    if (this.camera) {
      console.log('takePicture');
      const options = { quality: RNCamera.Constants.VideoQuality['1080p'], maxDuration: 0.5 };
      const record = await this.camera.recordAsync(options);
      console.log(record.uri);
      RNThumbnail.get(record.uri).then((result) => {
        console.log(result.path); // thumbnail path
        this.setState({ image: result.path });
        const { timer } = this.state;
        if (timer > 0) {
          setInterval(() => {
            this.takePicture();
          }, 5000);
        }
      });
    }
  };
```

n1ru4l, 2018-12-12:
This issue was originally related to `RCTCamera` which is deprecated. Please try to use `RCTCamera` instead. If you still experience any issues, please search for similar issues first before opening a new one.

closed 2018-12-12 by n1ru4l