react-native-camera/react-native-camera #418: [Android] Square like image ratio / aspect after image is taken

timoncp, 2016-09-20:
Hey man, managed to implement the camera fairly easily. Good job on this!

My only issue is that on Android, after taking a picture, it is cropped to a ratio that looks like on the iPhone for some reason. This happens even though beforehand, in the viewfinder, the entire screen is filled. Is there any way to control the ratio/size of the outputted picture?

Note: I have tried all different aspect constants and they all seem do to the same thing.

### Steps to reproduce
1. Take Photo
2. Open photo

### Expected behaviour
Image is full screen size/ratio after picture is taken.

### Actual behaviour
Image is cropped to a square like ratio after picture is taken.

### Environment
- **Node.js version**: 4.4.3
- **React Native version**: 0.33
- **React Native platform + platform version**: Android 5.0

### react-native-camera
**Version**: master

diegocouto, 2017-04-21:
Hi, @timoncp! 

We're currently facing the same problem. Have you managed to find a fix to this issue? 😄 

Thanks!

timoncp, 2017-05-10:
Glad you figured it out, @diegocouto . I haven't been active on the React Native side for a while now, so doubt I could've been of any help :( . Was thinking of suggesting another library or Expo.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco