react-native-camera/react-native-camera #419: Limit onBarCodeRead detection rate?

kevinnguy, 2016-09-22:
Is it possible to limit `onBarCodeRead` detection rate? Currently, when I scan over the same barcode for a period of time, `onBarCodeRead` is fired multiple times. 

In my use case, I'd like to be able to detect the barcode on screen and call `onBarCodeRead` only once, while the barcode is still on screen.

michaelswe, 2016-09-25:
You could always throttle the call to onBarCodeRead use [https://lodash.com/docs/4.16.1#throttle](url) or maybe  [https://www.npmjs.com/package/core-decorators](url)

phoenixbox, 2016-09-28:
@kevinnguy @forrest23 like @michaelswe said, a throttle does what you want. Heres a code snippet 🚀 

```
import React, {
  Component
} from 'react';
import {
  View
} from 'react-native';
import _ from 'lodash';

class YourCameraComponent extends Component {

  constructor(props) {
    super(props);

    /*
      Setup your onBarCodeRead throttle here
    */
    this.onBarCodeRead = _.throttle(this.onBarCodeRead, 500);
  }

  render() {
    const previewStyles = {
      flex: 1,
      justifyContent: 'flex-end',
      alignItems: 'center',
    }

    return (
      <View style={{flex: 1}}>
        <Camera ref={(camera) => { this._camera = camera; }}
          style={previewStyles}
          onBarCodeRead={this.onBarCodeRead.bind(this)}
          {...yourCameraPropsHere} />
      </View>
    );
  }

```

michaelswe, 2016-09-28:
@kevinnguy If you really want it to be called only once, and if you're using Lodash anyway, you could use [https://lodash.com/docs/4.16.2#once](url)

I try to limit my dependencies in my projects, but I'd have a hard time doing without Lodash. Lodash is very mature and stable, so I wouldn't hesitate to depend on it, even if the task at hand is a simple call counter for limiting function calls.

phoenixbox, 2016-09-28:
@michaelswe didn't know about once 🙌

kevinnguy, 2016-09-28:
Thanks guys for the great advice! @phoenixbox yeah, I did exactly that but in my `onBarCodeRead` method, I also call lodash's `cancel` to cancel throttling if it's scanning the same barcode more than once. 

@michaelswe that's pretty cool! I'll give it a try!

closed 2016-09-28 by kevinnguy

adiii4, 2018-09-15:
I tried to throttle the call on onBarCodeRead with 3000ms but it doesn't work as expected. I hold the barcode in front of the camera and when the barcode is read (the phone vibrates when that happens), I **immediately** move the barcode away from the camera. Then after about 3 seconds the same barcode gets read again (phone vibrates again) even though there is no barcode in front of the camera? Why does that happen?

sinkeat, 2018-10-01:
> I tried to throttle the call on onBarCodeRead with 3000ms but it doesn't work as expected. I hold the barcode in front of the camera and when the barcode is read (the phone vibrates when that [...]
Hi adiii4, i have faced same issue before, i found that is caused by default trailing true of lodash.throttle, may try:
`this.throttledOnBarCodeRead = lodash.throttle(this.onBarCodeRead, this.props.scannerCooldown, {trailing:false});
`

almazpdf, 2019-08-09:
Did trailing false resolve the issue? Has anyone tried debounce instead?