react-native-camera/react-native-camera #421: Cloning Project, Running Example Doesn't Work in Xcode 8

timcamber, 2016-09-23:
### Steps to reproduce
1. Clone project
2. Follow instructions in `Example/README.md`

### Expected behaviour
Example app should run in Xcode 8

### Actual behaviour
Build errors occur when trying to build in Xcode 8

### Environment
- **Node.js version**: `6.5.0`
- **React Native version**: `0.26.3`
- **React Native platform + platform version**: iOS `10.0.2 (14A456)`, Xcode `8.0 (8A218a)`

### react-native-camera
**Version**: on `master`, commit `1747ec4b0a52472cf2e8ad95f2c3603595e28e13`

charpeni, 2016-09-23:
Thank you for reporting, i'm working on this right now.

atinder90, 2016-09-28:
In my case build errors are gone after updating react and react-native but app crashed while rendering <Camera />
Using Xcode 8

getnashty, 2016-09-29:
+1 @atinder90

royisch, 2016-10-01:
@charpeni - any updates? my app stopped working with the camera once i upgaraded xcode. i get EXC_BAD_ACCESS once accessing the camera

muratozgul, 2016-10-03:
see #386 (nitrog7's answer solved my issue)

PR #422 (2016-11-18): Update Example to RN 0.34 and Xcode 8
Fixes #421.

cc @rt2zz
closed 2016-11-18 by charpeni