react-native-camera/react-native-camera #426: Crash on second invocation

michaelswe, 2016-09-26:
### Steps to reproduce
1. Open a view with react-native-camera
2. N.B. Not necessary to capture picture
3. Close view
4. Open view again

### Expected behaviour
I'd expect the same behavior as at the first invocation.

### Actual behaviour
Crash.

    - (void)setCaptureQuality:(NSString *)quality
    {
        if (quality) {
            [self.session beginConfiguration];
            if ([self.session canSetSessionPreset:quality]) {
                self.session.sessionPreset = quality;
            }
            [self.session commitConfiguration];
        }
    }

At *[self.session commitConfiguration]* XCode reports "EXC_BAD_ACCESS"

### Environment
- **Node.js 6.4**:
- **React Native 0.33**:
- **iOS 10.0, simulator

### react-native-camera
**Version**: lwansbrough/react-native-camera  (6074ec3)
*This is the version with the simulator fix*

furkancelik, 2016-09-26:
I'm having the same problem too...!

michaelswe, 2016-09-26:
N.B. the problem only exists in the simulator. Works fine on device.

JoschaP, 2016-09-27:
same problem here

andrerfneves, 2016-09-27:
This is relating to iOS 10.0. I fixed the issue with this inside Info.plist

``` 
<key>NSCameraUsageDescription</key>
<string>This app uses the camera.</string>
```

The string can really be anything you want, it'll display when the app launches and asks for permissions to use the camera.

ibussieres, 2016-09-27:
@andrerfneves it does not fix for me. The second invocation crashes. I have the the NSCameraUsageDescription key set in my .plist files.

andrerfneves, 2016-09-27:
@ibussieres I didn't run into second invocation issues. But whenever I flipped the camera to the front camera the app crashed. I had to add the same key and string pair on the .plist file but for the Microphone native properties.

```
<key>NSCameraUsageDescription</key>
<string>This app uses the camera.</string>
<key>NSMicrophoneUsageDescription</key>
<string>This app uses the microphone.</string>
```

Not sure if this will fix your issue, but this is what helped me solve it.

michaelswe, 2016-09-28:
As this only happens in the simulator I'm guessing that there is something more that needs to be wrapped in an `#if !(TARGET_IPHONE_SIMULATOR)` _(see 6074ec3)_. Maybe there is something that should be initialized only once per session. Unfortunately I don't know Objective C, so I can't look into it.

JoschaP, 2016-09-28:
@andrerfneves it does not fix it for me, too.

I've added both to the plist file:

<key>NSCameraUsageDescription</key>
<string>This app uses the camera.</string>
<key>NSMicrophoneUsageDescription</key>
<string>This app uses the microphone.</string>

breaks at line: 988 of RCTCameraManager
[self.session commitConfiguration];

2016-09-28 14:16:48.562 [info][tid:main][RCTCameraManager.m:270] Error setting property 'captureAudio' of RCTCamera with tag #342: capturing audio

ibussieres, 2016-09-28:
@andrerfneves thanks for the hinters, unfortunately it does not work. Besides, I am using this package only for barcode scanning so I had disabled the microphone. I think @michaelswe has put the finger on the type of problem that we need to surface up, I'm not too good with Objective-C either, but I'll try to look into something (yet hoping someone with better Objective-C skills will, too ;) )

chrisnojima, 2016-09-28:
I have the same problem. It renders fine but if I switch screens and come back it'll crash. I'm running the iPhone 7 iOS 10.0 simulator. I have only the NSCameraUsageDescription as i have autoCapture turned off (and changing the microphone setting seems to change nothing regarding this issue).

For me it crashes at: https://github.com/lwansbrough/react-native-camera/blob/master/ios/RCTCameraManager.m#L988

[image]

comountainclimber, 2016-09-28:
I am experiencing a crash at the same place @chrisnojima have permission in info.plist for both camera and microphone... 

This crash appears to only occur on simulator as others have stated above

miclaus, 2016-10-01:
+1 also have permissions in Info.plist

shahen94, 2016-10-14:
@michaelswe try to put into `<Camera />` component empty <View /> .
Like this.

```
<Camera>
  <View />
</Camera>
```

RGBz, 2016-10-15:
@shahen94 did that work for you? I'm still getting a crash at the same line.

shahen94, 2016-10-18:
@RGBz It works for me

rgoldiez, 2016-11-04:
Added a `<View />` inside of `<Camera>` and it still crashes in the simulator on the second invocation.

comountainclimber, 2016-11-05:
(this issue only occurs in simulator)

uncle-T0ny, 2016-11-11:
I have crash on setCaptureQuality and this bug reproduces on real device. 
I tried to use different component properties, but this has not any effect.
[image]

[IOS camera crash.txt](https://github.com/lwansbrough/react-native-camera/files/586116/IOS.camera.crash.txt)

And I don't know how to solve it, could anyone help me, please?

baskan, 2016-11-11:
I dunno if its releated but I put camera inside a wrapping container and add a shouldComponentUpdate() { return false; } now working fine on actual android device havent tested on an actual iOS device but its worth a try.

AndrewDeveloper, 2016-11-11:
Not sure how and why it works, but @shahen94 fix helped me

kesha-antonov, 2016-11-12:
[image]

Got this too. Sadly I don't have enough knowledge for now to fix this(

Hoping you can fix this guys.

uncle-T0ny, 2016-11-13:
@kesha-antonov I've just fixed the issue. The problem really was with NSMicrophoneUsageDescription permission. Check that application has camera and microphone permissions before you render the Camera component. Also be aware that microphone permission request not work on iPhone simulators. Good luck to you.

kesha-antonov, 2016-11-13:
@uncle-T0ny Thanks!
And maybe ```captureAudio={false}``` will be good when use camera only for photos. Hence do not need microphone permissions.

rgoldiez, 2016-11-13:
I did not add the microphone permission because I set video to false. I use it just for still images. No crashes here.

jqn, 2016-11-16:
Same here I have captureAudio={false} because I don't need to capture audio. Everything works fine in iOS but the Android version crashes.

closed 2016-11-18 by charpeni

Okipa, 2017-05-11:
I am experiencing the same issue this issue on Android with the v0.6.
I have specified `captureAudio={false}` and this is still crashing on Android.
I tried to add `<uses-permission android:name="android.permission.RECORD_AUDIO" />` on my AndroidManifest.xml file and the crash still happens.
Is there a way to avoid this problem ?

Geforceyu, 2017-09-05:
I'm having the same problem too...! I add this key to info.plist ,was Iphone