react-native-camera/react-native-camera #430: [!] No podspec found for `react-native-camera` in `../node_modules/react-native-camera`

lotosbin, 2016-09-28:
follow REAMD.md
add `pod 'react-native-camera', path: '../node_modules/react-native-camera'`
when run `pod install` get :
```
[!] No podspec found for `react-native-camera` in `../node_modules/react-native-camera`
```

miss *.podspec file in npm registry.

michaelswe, 2016-09-28:
I find pods hard to work with and try to stay away from them. react-native-camera is not pod-dependent. Have you tried the simpler way: `react-native link react-native-camera`

Anyway. I had a similar problem with a pod (mixpanel). The problem was the header search path. I'm still not sure what the base for these relative paths are, but I added more paths to be sure, and it worked.

Docs says that the header paths are relative to the folder with the xcode project resides. But is it the pod project file or is it the folder where my workspace resides? Anyway, my Mixpanel pod works with these search paths. I should investigate which of them actually is needed, but I have other things to do..

    $(SRCROOT)/../react-native/React
    $(SRCROOT)/../../react-native/React

In the xcode project file for the project, under Build Settings. Search for **Header Search Paths** and make sure they point to a reasonable place.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco