react-native-camera/react-native-camera #432: Auto-focus doesn't work on back camera

unipheye, 2016-09-28:
### Expected behaviour
I expect the back camera to auto-focus for a sharp view.

### Actual behaviour
I have created a simple React Native app with react-native-camera just for testing. All works fine except the back camera doesn't auto-focus. I have tried to simplify the script just to be sure no settings are interfering but without any good result.

### Environment
- **Node.js version**: 5.10.1
- **React Native version**: 0.34.0
- **React Native platform + platform version**: Android 5.1.1

### react-native-camera
**Version**: npm

andrerfneves, 2016-09-28:
Pretty sure you have to implement these two methods:
https://github.com/lwansbrough/react-native-camera#onfocuschanged-event--nativeevent--touchpoint--x-y--

Like so:

```
_handleFocusChanged() { /***/ }
_handleZoomChanged() { /***/ }
```

And then make sure you have these props in your Camera component:

```
defaultOnFocusComponent
onFocusChanged={this._handleFocusChanged}
onZoomChanged={this._handleZoomChanged}
```

unipheye, 2016-09-28:
Hi @andrerfneves, I tried that but still nothing seems to happen. Tap-to-focus also doesn't work either. The view stays in the same focus area and doesn't adjust.

harshq, 2016-11-02:
Any luck with this ? Im having the same issue :/

kenmaca, 2016-12-04:
Also having the same issue here. iOS `onFocusChanged` works as intended.

abrahambotros, 2017-01-30:
Hi all, please check out https://github.com/lwansbrough/react-native-camera/pull/575 for tap-to-focus and improved (continuous) auto-focus. I've also noted in the README that indeed `onFocusChanged` and `onZoomChanged` don't actually get called on Android yet, though pinch-to-zoom seems to work (without any of my PR changes) and https://github.com/lwansbrough/react-native-camera/pull/575 should enable tap-to-focus and better auto-focus as you guys have been mentioning.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco