react-native-camera/react-native-camera #437: Flash and Torch mode not working for Video

larryranches, 2016-09-29:
Does flash and torch mode work when recording video? It seems like it's not working for specifically only video but it does work for "still" or picture mode. Is this feature supported for video or a bug?

Any help would be much appreciated.

### Steps to reproduce
1. Set flash or torch mode to on
2. Set camera to video
3. Flash nor torch is working when shooting video but is working for stills.

### Expected behaviour
Flash and torch mode should work when Camera is in video mode

### Actual behaviour
Flash and torch mode not working when using Video

### Environment
- **Node.js version**: 6.6.0
- **React Native version**: 0.33
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc : IOS 10

### react-native-camera
**Version**:  master

larryranches, 2016-12-08:
@lwansbrough Any ideas on this?

flybayer, 2017-03-01:
Not working for me either with video mode on iOS!

Kkhattab, 2017-03-13:
Same here, would love a follow up on this issue.

brunsy, 2017-03-26:
Until this is fixed setting the torchMode to 0 or 1 should work. `torchMode={this.state.flashMode ? 1 : 0}`

larryranches, 2017-04-12:
bumpppp

mesbahsabur, 2017-05-01:
I found that the issue was caused by using a button which i positions 'absolute' over the camera component. Solved this by setting the camera as absolute and the buttons as regular views, like this:

Hope it helps!

```
render() {
    return (
      <View style={styles.container}>

        <View style={styles.cameraContainer}>
          <Camera
            torchMode={this.state.torch ? Camera.constants.FlashMode.on : Camera.constants.TorchMode.off}
            ref={(cam) => { this.camera = cam; }}
            // rest
          >
          </Camera>
        </View>

        <View style={styles.buttonContainer}>
          <TouchableOpacity onPress={() => this.onTorchButton()}>
            <View style={styles.button}>
              {this.state.torch ? <Icon color='white' type='material-community' name='flashlight-off' /> :
              <Icon type='material-community' name='flashlight' /> }
            </View>
          </TouchableOpacity>
        </View>

      </View>
    );
  }

const styles = StyleSheet.create({
  container: {
    flex: 1,
  },
  buttonContainer: {
    flex:1,
    padding:15,
    flexDirection: 'row',
    alignItems:'flex-end',
    justifyContent:'center'
  },
  cameraContainer: {
    position: 'absolute',
    width: width,
    height: height,
    top:0,
    left:0
  },
  button: {
    margin:20,
    flex:0,
    borderRadius:25,
    width:50,
    height:50,
    alignItems:'center',
    justifyContent:'center',
    backgroundColor:'red'
  },
});
```

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

cristianocca, 2019-05-10:
Still an issue, one year after: https://github.com/react-native-community/react-native-camera/issues/2171