react-native-camera/react-native-camera #440: Can resize view

npomfret, 2016-09-30:
Using latest rn-camera with RN0.34.1

I have a small camera view in my component which I want to make full screen if the users presses a button.  I'm using a <Modal> for the full screen bit, but the camera view that gets displayed inside it doesn't re-size.  It stays the same size as it was when being displayed as a small camera view.

How can I get the <Camera> to re-size itself.?

highbass, 2017-03-09:
@npomfret did you ever figure out how to solve this issue?

npomfret, 2017-03-09:
No sorry.

npomfret, 2017-03-20:
I'd love to get a solution to this. Anyone got any ideas?

The camera view size appears to be fixed, once you've created a `<Camera` any others you have in your app will be the same size, regardless of the style props you pass to it.

npomfret, 2017-03-21:
I think part of my problem, though still not fixed, is that I was showing the camera in one size, then pushing another view onto my stack in order to show the camera at a different size. So in effect I had 2 cameras open.  However, if I make sure I close the _hidden_ camera before opening the new one it still doesn't work - just get a blank screen.

XaviSmith, 2017-08-28:
@npomfret  what did you end up doing?

npomfret, 2017-08-28:
I failed to fix it.

XaviSmith, 2017-08-28:
@npomfret  Probably not relevant to you anymore but i found a sort-of fix
You can set the <Camera>'s height and width in style and change the aspect to fit or stretch like so

  <Camera
          ref={(cam) => {
            this.camera = cam;
          }}
          style={{height:200, width: 200}}
          captureTarget={Camera.constants.CaptureTarget.disk}
          aspect={Camera.constants.Aspect.fit}>
<Camera>

jolinkomo-code, 2017-09-04:
@XaviSmith tried that. That does not work unfortunately

mun3, 2018-06-02:
@XaviSmith's method crashes the Android emulator (API 26).

mun3, 2018-06-02:
@sibelius I'm using RCTCamera. I'll give RNCamera a try first.

mun3, 2018-06-02:
@sibelius The same issue applies to RNCamera and mater branch.

mun3, 2018-06-03:
In fact, my take is that the view size cannot be property adjusted using RNCamera because `Camera.constants.Aspect.fit` would be needed to fit the preview inside the smaller box. Am I right?

rndevzaki, 2020-05-06:
containerStyle={{width:'30%',height:'30%'}}
        cameraStyle={{width:'30%',height:'30%'}}

JaosnHsieh, 2020-11-05:
```jsx
 <RNCamera
        ref={(ref) => {
          this.camera = ref;
        }}
        style={{
          height: Dimensions.get('window').height
        }}
/>
```

works for me

Solaimanx, 2021-11-14:
@JaosnHsieh Thanks, Bro 
it also worked for me