react-native-camera/react-native-camera #442: Crash on Android (barcodeScannerEnabled)

saeho, 2016-10-03:
Using...
React Native 0.33
React-Native-Camera 0.3.8 (and also install from git url)

The app crashes with the following error:

> no propType for native prop RCTCamera.barcodeScannerEnabled

I was able to make the app work and eliminate the crash by deleting the following lines from RCTCameraViewManager.java

```
  @ReactProp(name = "barcodeScannerEnabled")
    public void setBarcodeScannerEnabled(RCTCameraView view, boolean barcodeScannerEnabled) {
        view.setBarcodeScannerEnabled(barcodeScannerEnabled);
    }
```

rcugut, 2016-10-28:
I think the barcode scanner was added to Android on Sep 17, with pull request [#413](https://github.com/lwansbrough/react-native-camera/pull/413).

@saeho  Are you using a version (git commit) after or before this?

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco