react-native-camera/react-native-camera #443: Wrong orientations when rotating the device

ahaus, 2016-10-03:
### Steps to reproduce
I used the CameraComponent as provided with the example and it looks fine in portrait mode.
However if I rotate the device (landscape) the camera/view finder shows a wrong orientation.
Even if I switch back to portrait mode it shows a wrong orientation (not correct portrait anymore)

### Expected behaviour
If I rotate the device from portrait mode to landscape mode the camera should show the preview in proper orientation (like it was in portrait mode) 

### Actual behaviour
When I rotate the device from portrait mode to the left in landscape mode the preview is rotated in the wrong direction (90 degrees instead of -90 degrees). However the image/photo taken is in the correct orientation.

### Environment
- **Node.js version**: 
6.6.0
- **React Native version**: 
0.34
- **React Native platform + platform version**: iOS 9.0, Android 5.0, etc
ios 9.3.5

### react-native-camera
**Version**: npm version or "master"
master

npomfret, 2017-04-26:
@ahaus  are you able to test my PR?  It may help with this https://github.com/lwansbrough/react-native-camera/pull/682.

JasonMore, 2017-05-05:
@npomfret I just tried your branch https://github.com/lwansbrough/react-native-camera/pull/682 and it did not fix it for me. The odd thing is sometimes if I let it sit with the camera open, it will rotate itself to the correct position. 

Here is a youtube video of the issue: https://youtu.be/MJyzPgtjEb8

npomfret, 2017-05-08:
Ah, that's a different bug - not what I was intending to fix.  The issue I was concerned with was unnecessary rotation of saved images.

akriger, 2017-12-07:
+1.... I have this same over a year later

JLLLinn, 2017-12-24:
Problem is still there, even with latest version 0.12.0. What's interesting is that it only happens on iPad, my iPhone works fine when rotates

Nauzer, 2018-03-01:
Still there... on 1.0.1 even :(

Nauzer, 2018-03-01:
When running the RNExample app I don't have the same issue... can't seem to figure out what is the difference.

sibelius, 2018-03-01:
Try to do a clean install

closed 2018-03-01 by sibelius

BoylesWhite, 2019-07-04:
Still got this issue on a project - anyone have a fix?

nabidefacto, 2019-09-02:
+