react-native-camera/react-native-camera #446: Camera mode only & captureAudio={false} but still prompts for microphone permission

rgovindji, 2016-10-05:
In the iOS 9.3 simulator it works as expected but when I run my app on my device (iOS 10, iphone 6+) I get prompted for both the camera + microphone. 

### Steps to reproduce
1. Set  captureAudio={false} captureMode={Camera.constants.CaptureMode.still}
2. Deploy to device with iOS 10, iPhone 6+
3. Open camera view
4. Accept prompt for camera access
5. Prompted for microphone access as well

### Expected behaviour
The prompt for microphone access should not occur

### Actual behaviour
Prompted for microphone access on my actual device but not in the simulator

### Environment
- **Node.js version**: v5.10.0
- **React Native version**:v0.28.0
- **React Native platform + platform version**: iOS 10 + iPhone 6 plus

### react-native-camera
**Version**:  master - most recent commit 6074ec3c8275ef5eb5fc39489c9a379f6ced923b

levitanong, 2016-10-07:
+1 I have this as well.

```
Error setting property 'captureAudio' of RCTCamera with tag #8: capturing audio
2016-10-07 17:12:00.614457 TestApp[5123:3523918] Error setting property 'captureAudio' of RCTCamera with tag #8: capturing audio
```

jonzee, 2016-10-10:
+1 more over running camera on simulator (iPhone 6, iOS 10) shows in debug msg: `capturing audio`

roszelj, 2016-10-26:
+1 I have this as well.

ahmetabdi, 2016-10-26:
The app's Info.plist must contain an NSMicrophoneUsageDescription key with a string value explaining to the user how the app uses this data. +1

rgovindji, 2016-10-26:
@ahmetabdi I think that's unrelated to this issue, we don't want to ask for microphone access because we aren't using it. Unless you are saying that setting NSMicrophoneUsageDescription makes the access prompt go away?

stoshfabricius, 2016-11-23:
+1, seems to run fine on the simulator without asking for microphone permissions but crashes on real device ( iPhone 6s, iOS 10)

xwartz, 2016-12-01:
+1 I have some issue

MatthiasDebaere, 2017-02-27:
+1, like @rgovindji said, we don't want to ask for microphone access when we're not using it.

yogiben, 2017-02-27:
+1 We've got one star reviews because of this

bharath-kamath, 2017-02-28:
@MatthiasDebaere @yogiben 

Instead of `checkDeviceAuthorizationStatus` have you tried `checkVideoAuthorizationStatus`? This seems to just prompt for Camera Access and not the microphone.

manishtpatel, 2017-03-09:
@bharath-kamath 's solution works, it should be clarified in the doc that checkDeviceAuthorizationStatus is for both audio and video, as video typically considered audio and visual together.

truongthanh1303, 2017-10-03:
@bharath-kamath you save my day, thanks a lot :)

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco