react-native-camera/react-native-camera #447: Set resolution for videos?

GeoffreyPlitt, 2016-10-05:
RN: 0.33
RNC: You guys don't have published NPM versions, but we're pinned to a commit from around Sept 24th.

We're capturing videos with this module and then uploading them. Our use case does not require full 1080p video, since users will be watching on their phones, so we'd like to set the upload resolution to 480p or 720p.

We can set the quality (low, medium, high) and see minor differences in the file sizes (4.7mb, 6.9mb, 7.1mb).

But what I can't figure out how to do, is set the resolution. In all 3 cases, the video is captured (and uploaded) at 1080p. Decreasing to 720p or 480p would likely improve our users's speed experiences on upload. We already have a CDN which modifies the resolution for download, but now that we're using your module, uploading is very slow since they're always 1080p.

--> How do we set the recorded resolution?

Related (but focused on images, not video): https://github.com/lwansbrough/react-native-camera/issues/222

GeoffreyPlitt, 2016-10-27:
Any ideas?

charpeni, 2016-11-18:
I'll close this issue since it was introduced by #469, if you think it should still be opened let us know.

closed 2016-11-18 by charpeni