react-native-camera/react-native-camera #450: Add Version Support Table

dangnelson, 2016-10-06:
I'm working on a standard for displaying version support on React-Native module README files. I would love your feedback: [react-native-version-support-table](https://github.com/dangnelson/react-native-version-support-table/)

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

sibelius, 2018-01-12:
@dangnelson this looks good, feel free to send a PR to solve this

ospfranco, 2018-01-12:
Looks like this last commit was 8 months ago