react-native-camera/react-native-camera #451: A way to read data byte from current preview, like a previewCallback?

kenma9123, 2016-10-08:
Is there a way to read the data from the current preview? My goal is to retrieve the camera preview as a Pixel Array.

A preview callback would be a good addition, passing the data byte, just like the onPreviewFrame callback when you set it up.

rameshvishnoi90904, 2016-10-19:
yes, need to preview the captured images

MathieuMailhos, 2016-11-12:
Agreed, need a preview callback function.

ospfranco, 2018-01-12:
Hey! this repository was recently transferred to the react-native-community, as part of our ongoing cleaning process we are closing old tickets (there is just too many of them!), please feel free to try the latest versions or the master branch to see if your problem has been solved.

If you are still experiencing this problem feel free to re-open this or fill a new ticket.

closed 2018-01-12 by ospfranco

DennisJu007, 2019-06-12:
@n1ru4l Has  this issue was fixed in new release version?How can i use it.

n1ru4l, 2019-06-12:
@DennisJu007 Please open a new issue or even better a pull request with an implementation.